openshwprojects/OpenBK7231T_App #711: implement some web ui security

ProgramadorUmGraytec, 2023-03-06:
Hello, it's not really a problem but an improvement, I'm a tasmota user and on my devices I can configure a username and password for when I enter the web ui, the BK7231N firmware does not have this function, would it be very difficult to implement?

(if I wrote something wrong I'm sorry, I'm using google translator because I'm BR)

vivalenta, 2023-07-11:
+1, need simple login/pasword protection

analystcmyk, 2023-08-16:
+3 really. Even when only basic auth.

analystcmyk, 2023-09-12:
Even without username would be ok, or a token to be included in the url, i just can't have my kids switch off the freezer just for fun :)

SetChannelVisible is not really an option because they will find out the web app  set the channel visible again.

They are savvy, but a brute force is not not yet in their toolkit I hope.

Of course they can just switch on/off 5 times to reset all.

castorw, 2023-12-13:
PR in progress: #996