openshwprojects/OpenBK7231T_App #712: Issues Update OTA from version 1.15.476 to lastest

Wolfieeewolf, 2023-03-07:
Have been trying to update my Arlec PC191 power plug with energy meter device for about a week now. I am using the Web UI and from what I can see it looks like it updates but on reboot the version number is still 1.15.476. I have tried using the OTA_http command and it looks like it is throwing some errors. 

Info:CMD:[WebApp` Cmd 'ota_http https://github.com/openshwprojects/OpenBK7231T_App/releases/download/1.15.559/OpenBK7231T_1.15.559.rbl' Result] OK
Info:HTTP_CLIENT:Parse url https://github.com/openshwprojects/OpenBK7231T_App/releases/download/1.15.559/OpenBK7231T_1.15.559.rbl
Info:HTTP_CLIENT:host: 'github.com', port: 80
Info:HTTP_CLIENT:HAL_TCP_Establish: created socket 2

Error:HTTP_CLIENT:success to establish tcp, fd=2
Info:OTA:init OTA, startaddr 0x132000
Info:OTA:
myhttpclientcallback state 0 total 0/0
Info:HTTP_CLIENT:httpclient_recv 167 bytes has been read
Error:HTTP_CLIENT:Could not found
Error:HTTP_CLIENT:httpclient_recv_response is error,ret = -43
Info:OTA:
Info:OTA:close OTA, addr 0x132000
Info:OTA:
myhttpclientcallback state 2 total 0/0
Info:OTA:Rebooting in 1 seconds...
Info:MQTT:Publishing val 245.975906 to obk42AA598E/voltage/get retain=0

When using the Web UI OTA it gives me this

Info:MAIN:Time 187, idle 202005/s, free 88488, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:OTA:init OTA, startaddr 0x132000
Info:OTA:132000
Info:OTA:141000
Info:OTA:142000
Info:OTA:143000
Info:OTA:144000
Info:MAIN:Time 190, idle 406622/s, free 72760, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:GEN:dhcp=0 ip=192.168.0.40 gate=192.168.0.1 mask=255.255.255.0 mac=*********** 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-36,ssid=**********,bssid=********** ,channel=******,cipher_type:CCMP
Info:OTA:145000
Info:OTA:146000
Info:OTA:147000
Info:OTA:148000
Info:OTA:149000
Info:OTA:14a000
Info:OTA:14b000
Info:OTA:14c000
Info:OTA:151000
Info:OTA:152000
Info:OTA:153000
Info:OTA:154000
Info:OTA:15a000
Info:OTA:15b000
Info:OTA:15f000
Info:OTA:160000
Error:CMD:cmd Status NOT found (args 0)
Error:GEN:CHANNEL_HasChannelPinWithRole: Channel index 64 is out of range <0,64)

Info:OTA:161000
Info:OTA:162000
Info:OTA:167000
Info:OTA:168000
Info:OTA:169000
Info:OTA:16c000
Info:OTA:16d000
Info:OTA:16e000
Info:OTA:16f000
Info:OTA:170000
Info:MAIN:Time 193, idle 199292/s, free 72760, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:OTA:175000
Info:MAIN:Time 194, idle 10875/s, free 72760, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:OTA:176000
Info:OTA:177000
Info:OTA:17a000
Info:OTA:17b000
Info:OTA:17c000
Info:OTA:17d000
Info:OTA:17e000
Info:OTA:17f000
Info:OTA:186000
Info:OTA:187000
Info:OTA:188000
Info:OTA:189000
Info:OTA:18a000
Info:OTA:18b000
Info:OTA:18c000
Info:OTA:18d000
Info:OTA:193000
Info:OTA:194000
Info:MAIN:Time 196, idle 237909/s, free 72544, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 4/38 
Info:OTA:195000
Info:OTA:196000
Info:OTA:197000
Info:OTA:198000
Info:OTA:199000
Info:OTA:19a000
Info:OTA:19b000
Info:OTA:19c000
Info:OTA:1a0000
Info:OTA:1a1000
Info:OTA:1a2000
Info:OTA:1a3000
Info:MAIN:Time 198, idle 450739/s, free 88360, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Module reboot in 2...

any ideas ??

openshwprojects, 2023-03-07:
Well, does the Web App OTA works? The HTTP ota is HTTP only right now, and Github link is HTTPS, so that is simply not supported yet. HTTPS would add more flash usage to the build.

Wolfieeewolf, 2023-03-07:
The web app doesn't work either.  all my other devices update fine. This one refuses to work. I flashed it with tuya convert initially and i haven't been able to update it since. I have flashed other devices they same way and had now issues. it's just this one device. Kinda stumped as to why it would be doing this. everything else on the device seems to run fine

Wolfieeewolf, 2023-03-07:
I've also tried 3 different browsers to see if that was the issue. I've cleared cache data as well  but still no luck

Wolfieeewolf, 2023-03-07:
there is nothing running. the only thing i have setup so far is the WiFi. I did clearconfig and clearall basically reset the whole device. No drivers running that I see.

Wolfieeewolf, 2023-03-07:
[image]

Wolfieeewolf, 2023-03-07:
[This is the device](https://www.elektroda.com/rtvforum/topic3944452.html)

openshwprojects, 2023-03-07:
I have never seen anything like that. It really does not update?

Can you setup a HTTP server (not HTTPS like github, ofc) and try the old HTTP OTA mechanism via URL? Do you know how to quickly setup HTTP server on your machine?

Wolfieeewolf, 2023-03-07:
I know it's weird!! 

It appears to update but then just doesn't. I have no idea.

I set up a HTTP server and tried updating with the Simple OTA system and also the ota_http command but still nothing

I get this on the logs

Info:CMD:[WebApp Cmd 'ota_http http://127.0.0.1:8080/OpenBK7231T_1.15.562.rbl' Result] OK
Info:HTTP_CLIENT:Parse url http://127.0.0.1:8080/OpenBK7231T_1.15.562.rbl
Info:HTTP_CLIENT:Host split into 127.0.0.1 port 8080
Info:HTTP_CLIENT:host: '127.0.0.1', port: 8080
Info:HTTP_CLIENT:HAL_TCP_Establish: created socket 2

Error:HTTP_CLIENT:connect error
Error:HTTP_CLIENT:fail to establish tcp
Error:HTTP_CLIENT:establish connection failed, error -1 
Error:HTTP_CLIENT:httpclient_connect is error,ret = -8
Info:OTA:
Info:OTA:close OTA, addr 0xff000
Info:OTA:
myhttpclientcallback state 2 total 0/0
Info:OTA:Rebooting in 1 seconds...
Info:MAIN:Time 258, idle 223571/s, free 84672, MQTT 0(16), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Main_Init_Before_Delay
Info:CFG:####### Boot Count 90 #######
Warn:CFG:CFG_InitAndLoad: Correct config has been loaded with 7 changes count.
Error:CMD:lfs is absent
Info:GEN:PIN_SetupPins pins have been set up.
Info:MAIN:Main_Init_Before_Delay done
Info:MAIN:Main_Init_Delay
Info:MAIN:Main_Init_Delay done
Info:MAIN:Main_Init_After_Delay
Info:MAIN:Using SSID [TelstraCD3B4A]
Info:MAIN:Using Pass [yb8v553y5h]
Info:MQTT:MQTT_RegisterCallback called for bT obk42593713/ subT obk42593713/+/set
Info:MQTT:MQTT_RegisterCallback called for bT bekens/ subT bekens/+/set
Info:MQTT:MQTT_RegisterCallback called for bT cmnd/obk42593713/ subT cmnd/obk42593713/+
Info:MQTT:MQTT_RegisterCallback called for bT cmnd/bekens/ subT cmnd/bekens/+
Info:MQTT:MQTT_RegisterCallback called for bT obk42593713/ subT obk42593713/+/get
Error:CMD:lfs is absent
Info:MAIN:Main_Init_After_Delay done
Info:MAIN:Time 1, idle 202892/s, free 93672, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 2, idle 201896/s, free 93672, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 3, idle 406076/s, free 93672, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 4, idle 205738/s, free 93672, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 5, idle 204320/s, free 93672, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:ssid:******* key:*******
Info:MAIN:wl_status 3
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING - 1
Info:MAIN:wl_status 6
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_DISCONNECTED - 2
Info:MAIN:wl_status 6
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_DISCONNECTED - 2
Info:MAIN:wl_status 10
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED - 4
Info:MAIN:wl_status 11
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED - 4
Info:MAIN:Time 6, idle 208798/s, free 89160, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Boot complete time reached (5 seconds)

Wolfieeewolf, 2023-03-07:
this is the Simple OTA system log

Info:HTTP:http_fn_ota_exec: will try to do OTA for http://127.0.0.1:8080/OpenBK7231T_1.15.562.rbl 
Info:HTTP_CLIENT:Parse url http://127.0.0.1:8080/OpenBK7231T_1.15.562.rbl
Info:HTTP_CLIENT:Host split into 127.0.0.1 port 8080
Info:HTTP_CLIENT:host: '127.0.0.1', port: 8080
Info:HTTP_CLIENT:HAL_TCP_Establish: created socket 2

Error:HTTP_CLIENT:connect error
Error:HTTP_CLIENT:fail to establish tcp
Error:HTTP_CLIENT:establish connection failed, error -1 
Error:HTTP_CLIENT:httpclient_connect is error,ret = -8
Info:OTA:
Info:OTA:close OTA, addr 0xff000
Info:OTA:
myhttpclientcallback state 2 total 0/0
Info:OTA:Rebooting in 1 seconds...

Wolfieeewolf, 2023-03-07:
I'd manually flash the device but it's tightly sealed and didn't want to have to bust it open if I didn't have to. I think that is what I'm up to. Maybe when I used tuya cloud-cutter something glitched out. 

The device works with the firmware it has now maybe I just need to leave it as that and not worry about updating it.

Will see what tomorrow brings for now I'm done with it and going to bed

Wolfieeewolf, 2023-03-07:
So after repeated tries at updating it finally just updated itself. No idea why or how. Very strange. Maybe updating with an HTTP server unlocked something and made the Web App OTA update work. 

Have reset it again to see if it reverts but it looks like everything is good for now.

Wolfieeewolf, 2023-05-02:
Reopening this as an issue again.

 I now have 5 of these devices and all of them are refusing to update. They all appear to update but it never actually does. All the logs state it has been updated. but somehow it reverts back to the old version on reboot. I have tried clearing the devices to defaults, doing the HTTP update and doing a normal update but it just will not work.

I tried using Tuya cloud-cutter as that is what I used to convert the devices to OpenBK. It finds the device but will not reflash it. 

the device is sealed shut and I don't want to open it to flash them if I can help it so again I'm stuck.

openshwprojects, 2023-05-02:
Is power save off?

update from which version to which version?

Wolfieeewolf, 2023-05-02:
powersave is off

updating one device from OpenBK7231T_1.15.624 and the other 4 from OpenBK7231T_1.15.688.

I tried downgrading the 4 devices to OpenBK7231T_1.15.624 to see if there was something about that version that allowed me to upgrade my first device. It was OpenBK7231T_1.15.476 and I'm not sure how but I eventually got it to update to OpenBK7231T_1.15.624. Thought it might have been something in the code that made it update but none of the devices will downgrade to that version so I'm guessing it was just luck.

I have completely reset the device to stock OpenBK. No driver, no autoexec.bat just barebones firmware with only the WIFI configured. 

I even tried it without the wifi configured by using two WIFI dongles so i could access the Web App but still no luck. 

have been trying to update with multiple different browsers, computers, phones, and tablets. Have also tried different versions of the firmware to see if maybe one might trigger it to update but after more then 30 different versions it still will not go.

I'm at a loss as to what to try next

DeDaMrAzR, 2023-05-02:
I tried on at least 2 different T devices and can not replicate that error - even downgraded to 1.15.467 and updated to the latest via web app.
[image]

[image]

Have you tried to power cycle the device right after OTA maybe?

Wolfieeewolf, 2023-05-02:
I think it's device specific as I don't have these issues with all my other devices. Seem to be an issue with the device itself as I have 5 of them and all of them will not update pass the initial firmware update. 

https://www.elektroda.com/rtvforum/topic3944452.html

Are there any logs or anything that might help that I can dump off the device? 

All the devices were converted using Tuya Cloud-Cutter. not sure if that is the issue either. Might have been something in the profile that didn't overwrite the flash fully. Seem odd that it would happen to 5 devices though.

DeDaMrAzR, 2023-05-02:
Can you try doing the update via flash tool? - https://github.com/openshwprojects/BK7231GUIFlashTool/releases/tag/v1.0.8

Is that the possibility? We have run into similar issue while testing but with a BK7231N device, not the T variant. Also downgrade is not recommended as the OTA space has been revised/redistributed and may lead to some undocumented errors.

Wolfieeewolf, 2023-05-03:
The only why to update via the flash tool is to open it to access the module. The device is ultrasonically welded shut and very hard to get open without damaging it in some way. I didn't want to open it if i didn't have to. 

Guess the device work well enough with the firmware it has now so I guess there is no real reason to update it. I would be purely updating just because there is on available.

Guess I'll just keep trying the updates as they happen and see if one of them will stick.

MnM001, 2023-05-06:
That's how I update my  Arlec Grid Connect PC191 devices:

1. Go to app (that's the web app built in the firmware)
2. Download the latest firmware (download it to your pc and browse to it when asked)
3. Apply the latest firmware
4. Wait for device to reboot
5. Press the button on the device and keep it pressed
6. Pull the device out from the wall power socket while keeping the button pressed
7. Wait 5 seconds and keep the button pressed
8. Plug the device back into wall power socket and release the button
9. Wait for device to boot and see the new firmware

Wolfieeewolf, 2023-05-07:
Excellent!

Have managed to update my devices to the latest firmware.

Would like to know why it only updates when you hold the button down.  Power still held on the flash module maybe?? 

At least we worked out how to update them. It's been bugging me for ages that I could work it out.

Wolfieeewolf, 2023-08-18:
Closing as issue resolved

closed 2023-08-18 by Wolfieeewolf