openshwprojects/OpenBK7231T_App #713: Interlock mode for smart relays

sodiniss, 2023-03-07:
Current setup:
addChangeHandler Channel1 == 1 backlog setChannel 2 0
addChangeHandler Channel2 == 1 backlog setChannel 1 0
This works, however this isn't perfect since **there still is a brief moment where both relays are on** and this causes problems on my roller shutter. A workaround would be managing these in home assistant, but I wonder if there's a script so that it turns off the other relay _before_ turning on the requested one.

Thanks in advance

openshwprojects, 2023-03-07:
Hello, what kind of driver do you need? I can write a dedicated (hardcoded), no-script solution just for you.

What is your use case? My shutters knowledge is not too well, I must admit, I haven't done shutters in my house yet.

Btw I think that you could also work around the problem with more advanced scripting. Something like... use 3 channels, 1 for control (as a bridge, as API) and two "internal" channels for relays. That would allow you to introduce a dead time to make sure never "both relays on" situation happen.

Smth like: Ch1 - input, Ch2 - relay 1, Ch3 - relay 3
addChangeHandler Channel1 == 1 backlog setChannel 2 0; setChannel 3 0; setChannel 2 1
addChangeHandler Channel1 == 0 backlog setChannel 2 0; setChannel 3 0; setChannel 3 1

sodiniss, 2023-03-07:
This looks promising, is there a way to add a little delay before the third backlog command? Like

addChangeHandler Channel1 == 1 backlog setChannel 2 0; setChannel 3 0; _wait 2ms_ ; setChannel 2 1

openshwprojects, 2023-03-07:
Hmm there could be multiple ways to do this, but our quick tick is 25ms and there is no elegant way to wait a shorter time than that. Would a 25ms delay be ok for you? If 25 ms is okay, then I can try to figure something out, but it require one little change to codebase.

sodiniss, 2023-03-07:
My bad, that was an example, actually I'd like it to be longer than that, like 300ms. I'm trying with  delay_ms 300 but doesn't looks good as of now.

openshwprojects, 2023-03-07:
delay_ms is supposed to be used in script threads. You can do that with a script thread or with a repeating event. The only issue is that, right now, repeating events are ticked every second.

Give me 15 minute and I will move that to quick tick, then you will need to do OTA update for your device, and then this will work:

addChangeHandler Channel1 == 1 backlog setChannel 2 0; setChannel 3 0; addRepeatingEvent 0.3 1 setChannel 2 1

syntax is: addRepeatingEvent [TimeSeconds] [TimeTorepeat]

sodiniss, 2023-03-07:
Awesome! I'll test this after lunch. If it works correctly in the future this could be a feature like Tasmota does with Interlock 1,2

openshwprojects, 2023-03-07:
How Interlock 1,2 works?

sodiniss, 2023-03-08:
Okay so after some testing this works, however it's still far from ideal since this virtual button approach is to be considered as a workaround. This does not solve the mutual exclusion problem since it just moves the dependancy on the new channel.

I think the best way to do this would be by overriding the "Turn on" function of the relay x  by adding a Turn off relay y + short delay _before_ the actual Turn on call. Like tasmota's Interlock. Would this be a possible new feature?

openshwprojects, 2023-03-10:
Yes, I will need to think how to handle it well.