openshwprojects/OpenBK7231T_App #714: BP5758D does not respect Flag24

no2chem, 2023-03-07:
**Describe the bug**

Hi All, I have a new Treatlife SL10 bulb which appears to be based on CB2L (BK7231N). I am able to get it to talk to the BP5758D on board (P7=BP5758D_CLK, P8=BP5758D_DAT), mapping (BP5758D_Map 4 3 2 1 0) but it turns out that it only has 4 channels, analyzing the board, channel 0 appears to be floating and the white LEDs are WW. 

Flag24 should cause CW to be emulated by RGB, but it doesn't seem to work with BP5758D. Instead the light sort of just dims out due to the empty channel.

**Firmware:**
- Version [e.g. 1.14.129] 1.15.559
- Device? Treatlife SL10 CB2L
- Chip/model: BK7231T? BK7231N
- Device config? P7=BP5758D_CLK, P8=BP5758D_DAT

**To Reproduce**
Configure BP5758D
Turn on flag 24
Set color temperature to 6000K

**Screenshots**

openshwprojects, 2023-03-10:
It's not a bug, it's a lack of feature. I will try to get it working for you today, can you help with testing?

openshwprojects, 2023-03-10:
I am not sure about this:
https://github.com/openshwprojects/OpenBK7231T_App/commit/add67dd280fea13d5c18bf1f6f712241f6fea62b
Can you check, what do you think?

no2chem, 2023-03-11:
Hm strange, I moved W to channel 4 (`BP5758D_Map 4 3 2 0 1`) but there's no W and it's now blue on cool white. At 2000K the bulb is now dark. 

I think at 6000K W needs to be at 100% on this bulb still

openshwprojects, 2023-03-15:
What may be the problem? I am not sure yet.

panchomira, 2023-04-03:
Hello! I have a similar problem, I think.

As far as I can tell, your bulb does not have a cool white LED. Therefore, the original Tuya firmware achieves it with RGB LEDs.

Currently, I am using Flag 24 ([LED] Emulate Cool White with RGB in device with four PWMS - Red is 0, Green 1, Blue 2, and Warm is 4), which tries to do the same thing, but it's way off because it doesn't work the same way as the original firmware.

In the original firmware, when you set the temperature to around 4500K, all the warm white LEDs and RGB LEDs turn on (although I don't think they're all at the same intensity). Then, when you select the coolest white, only the warm white LED and the blue RGB LED turn on, not the green or red.

I tested the white balance on the original firmware using a camera, and it's pretty accurate in terms of the Kelvin temperature. Even when emulating temperature with RGB, it produces very little color in the final light projection, staying very true to a pure white.

I would love to solve this issue because the temperature is not usable in the current state of OpenBk and this type of lamp. However, I have little to no knowledge in writing firmware.

If someone can help me with this, I am willing to spend some hours trying to calibrate it.