openshwprojects/OpenBK7231T_App #716: Easy way to identify sm2135 mapping?

lmcd, 2023-03-08:
Is there a non-invasive way of determining the sm2135 mapping of a bulb?
The default mapping kinda works, but the colours are in the right order, and the LED temperature is backwards.

Would be nice if the firmware provided a way of brute forcing combinations until it looks right to the user

openshwprojects, 2023-03-08:
Well, if you know that temperature is backwards, then the only possible solution is to swap C and W channels. If you then know that for example, blue and red are swapped, you also know 1:1 what to do... There is no need for guessing. If you know a current map (and you can know it, because u can always start with 0 1 2 3 4), then you will just need one more guess (or should I say attempt) to get it right. Just set red, blue, green, etc and observe what lights up.

lmcd, 2023-03-08:
Yeah I thought that, and tried swapping these numbers around in many combinations, but didn't get an accurate result :(

openshwprojects, 2023-03-08:
Okay, then... maybe I could add an automatic mechanism for that to Web App. That way it will work but it won't take much space in the firmware itself on the IoT device.

lmcd, 2023-03-12:
So I've noticed when the last 2 digits are `4 3`, the temperature and the brightness work correctly.

For the first 3 digits, I've tried all these combinations, but the colours still aren't lining up correctly:

0 1 2
0 2 1
1 2 0
1 0 2
2 0 1
2 1 0

I've also noticed that the brightness of any given colour appears to be around 50% of what the bulb looks like using the Smart Life app.

lmcd, 2023-03-12:
Actually, I've noticed that this is because the web ui is not sending the correct colours for red/blue/green (at least on a Mac). Red is sending as #ff2600 and not #ff0000, and when there's any mixing of channels going on, you get weird results.
Maybe a hue rotate is probably better for these bulbs than a full rgb colour picker.

Also, the brightness is still way off - I'm guessing we're not sending the right current value to the SM2135. It looks ok-ish at first glance, then you compare to an identical Smart Life-controlled bulb and the brightness is at least double.

lmcd, 2023-03-12:
Also, `SM2135_Current` only seems to affect CW. The command for me has no effect on RGB at all.