openshwprojects/OpenBK7231T_App #717: Set TuyaMCU channel as button

no2chem, 2023-03-09:
Hello,

I have a device with a TuyaMCU (TreatLife SS02) with a Nuvoton MCU. I wrote a custom Nuvoton MCU firmware that speaks TuyaMCU to OpenBK and passes all the LED switches and buttons to OpenBK as channels. 

It works well, I can set individual LEDs and send button change events over MQTT. However, I would like the device to handle button change events locally to determine whether to send a long press or short press, handling this in an homeassistant automation isn't great.

I looked into scripts but I don't think they quite accomplish what I want.

My TuyaMCU firmware handles all the debouncing already. It uses a bool DP and sends 1 when the button is pressed and 0 otherwise. Is there a way to send these channel button changes to the PIN handler for buttons, as if it were a physical GPIO on the OpenBK MCU?

no2chem, 2023-03-10:
I have no idea if any production devices are using it, most just have the TuyaMCU take over most of the controls. I don't think it'd be a serious change, but I haven't looked.

I'll probably post a repo quite soon, and after doing that I'll post to your forum. Don't need another gadget though 🙂.

openshwprojects, 2023-03-10:
Hmmm okay so... alright, I will do that for you. My proposition is that I can add a "virtual" GPIO that is non existent on Beken, something like, idk, GPIO30 and allow you to map a dpID to that GPIO. Then you'd use that GPIO as any other one. What do you think, would that suit your need?

no2chem, 2023-03-10:
Ah yeah, that would be perfect. Then I can also map the red led to "WIFI fail".

BTW, I pushed the repo up at
https://github.com/no2chem/open-tuyamcu

Will post on your forum later.

openshwprojects, 2023-03-10:
I will try to get it running for you fast, but I must triple-check how many possible GPIOs Beken chips have. I don't want to mark potential P29 as virtual while there can be a similar BK chip with physical P29...

Luckily our struct has allocated 32 slots for GPIOs:
[image]
so you'll get most like slots P29, P30, and P31 as virtual... so 3 pins.

no2chem, 2023-03-10:
Ah well that works out perfectly for my case. Yeah don't mean to cause anything urgently but I can probably test it out quickly.