openshwprojects/OpenBK7231T_App #718: Btn_Tgl_All Fails to work after two presses

propgfx, 2023-03-09:
I have a WiFi smart strip that I have flashed with the latest firmware.
There is a single button on the device which I have programmed to toggle all relays.
It works for two cycles and then fails to change the state of the relays to anything but off.

**Firmware:**
- Version 1.15.572
- Device: Smart WiFi Strip
- Chip/model: BK7231N
- Device config?
  "pins": {
    "6": "Rel_n;2",
    "7": "Btn_Tgl_All;0",
    "8": "Rel_n;1",
    "10": "Rel_n;6",
    "14": "Rel;5",
    "21": "WifiLED;0",
    "23": "Rel_n;1",
    "24": "Rel_n;3",
    "26": "Rel_n;4"
  },

**To Reproduce**

Here are the logs from when I press the button four times, on the first two presses it works as expected but teh third and fourth always set the channels to zero (main and mqtt logs disabled)
This is imediately after a restart and all relays are off.
If relays are set to on then a press of the button always turn them to off once this state has been reached.
If all the relays are set to on after a restart then pressing teh button toggles them all off, another press toggles them off and no change after that.

Logs: (all relays off to start with)

Info:GEN:7 Button_OnInitialPressDown
Info:GEN:7 key_short_press
Info:GEN:CHANNEL_Set channel 0 has changed to 1 (flags 0)

Info:GEN:CHANNEL_Set channel 1 has changed to 1 (flags 0)

Info:CFG:####### Flash Save Channel 1 as 1 #######
Info:GEN:CHANNEL_Set channel 2 has changed to 1 (flags 0)

Info:CFG:####### Flash Save Channel 2 as 1 #######
Info:GEN:CHANNEL_Set channel 3 has changed to 1 (flags 0)

Info:CFG:####### Flash Save Channel 3 as 1 #######
Info:GEN:CHANNEL_Set channel 4 has changed to 1 (flags 0)

Info:CFG:####### Flash Save Channel 4 as 1 #######
Info:GEN:CHANNEL_Set channel 5 has changed to 1 (flags 0)

Info:CFG:####### Flash Save Channel 5 as 1 #######
Info:GEN:CHANNEL_Set channel 6 has changed to 1 (flags 0)

Info:GEN:7 Button_OnInitialPressDown
Info:GEN:7 key_short_press
Info:GEN:CHANNEL_Set channel 2 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 2 as 0 #######
Info:GEN:CHANNEL_Set channel 1 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 1 as 0 #######
Info:GEN:CHANNEL_Set channel 6 has changed to 0 (flags 1)

Info:GEN:CHANNEL_Set channel 5 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 5 as 0 #######
Info:GEN:CHANNEL_Set channel 1 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 1 as 0 #######
Info:GEN:CHANNEL_Set channel 3 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 3 as 0 #######
Info:GEN:CHANNEL_Set channel 4 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 4 as 0 #######
Info:GEN:7 Button_OnInitialPressDown
Info:GEN:7 key_short_press
Info:GEN:CHANNEL_Set channel 2 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 2 as 0 #######
Info:GEN:CHANNEL_Set channel 1 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 1 as 0 #######
Info:GEN:CHANNEL_Set channel 6 has changed to 0 (flags 1)

Info:GEN:CHANNEL_Set channel 5 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 5 as 0 #######
Info:GEN:CHANNEL_Set channel 1 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 1 as 0 #######
Info:GEN:CHANNEL_Set channel 3 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 3 as 0 #######
Info:GEN:CHANNEL_Set channel 4 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 4 as 0 #######
Info:GEN:dhcp=0 ip=192.168.0.200 gate=192.168.0.1 mask=255.255.255.0 mac=10:5a:17:8d:0f:08 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-30,ssid=________,bssid=10:27:f5:3d:56:46 ,channel=1,cipher_type:CCMP
Info:GEN:7 Button_OnInitialPressDown
Info:GEN:7 key_short_press
Info:GEN:CHANNEL_Set channel 2 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 2 as 0 #######
Info:GEN:CHANNEL_Set channel 1 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 1 as 0 #######
Info:GEN:CHANNEL_Set channel 6 has changed to 0 (flags 1)

Info:GEN:CHANNEL_Set channel 5 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 5 as 0 #######
Info:GEN:CHANNEL_Set channel 1 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 1 as 0 #######
Info:GEN:CHANNEL_Set channel 3 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 3 as 0 #######
Info:GEN:CHANNEL_Set channel 4 has changed to 0 (flags 1)

Info:CFG:####### Flash Save Channel 4 as 0 #######

openshwprojects, 2023-03-10:
Thanks, I will look into that this morning.

openshwprojects, 2023-03-10:
There was indeed a bug which manifested only in certain case (where you used channels from 1 for relays but kept channel 0 for button...). It's now fixed, can you check?

propgfx, 2023-03-10:
Perfect!
It works great, thank you.
I will do a teardown of this device today and post.

openshwprojects, 2023-03-10:
Thank you, I am looking forward to reading the rteardown.