openshwprojects/OpenBK7231T_App #728: Migrating from LibreTuya (OTA)

esnet146, 2023-03-14:
There is an option to migrate by air from Openbeken to LibreTuya
https://docs.libretuya.ml/docs/flashing/esphome/#migrating-from-openbeken-ota

Is it possible to create a file in uf2 format for openbeken firmware? Any version for BK7231T and BK7231N chips
Description format https://docs.libretuya.ml/docs/ota/

openshwprojects, 2023-03-14:
I must admit that I am not sure if it's ethical to create a migration guide from another firmware, I don't know who creates articles like "migrating from OpenBeken ota", it's like... well... very aggressive advertising, I condemn that, but to answer your question, the topic has popped on our forum at least once:
https://www.elektroda.com/rtvforum/topic3958492.html
but from my knowledge, the command you need to use, in order to create UF2 format with OBK is:

ltchiptool uf2 write -f bk7231t ;;download;file.rbl

so you basically:
1. download RBL from us
2. use it to create UF2 with command above
3. feed UF2 to libretuya
4. your device will reboot to OBK....

I hope it answers your question, feel free to ask if it's still not working, but I don't know much about their ecosystem I am surprised that they actually took their time to create "how to abandon obk" guide. Huh.

closed 2023-03-14 by openshwprojects

esnet146, 2023-03-15:
ltchiptool uf2 write -f bk7231u -b generic-bk7231t-qfn32-tuya -o openbkt.uf2 'download;OpenBK7231T_1.15.440.rbl'
Thanks!!!

kuba2k2, 2023-03-18:
@openshwprojects I was wondering when will you notice that. How can you call it advertising, if it's in my own docs page?

I added that section to the docs, because people were asking about that frequently, and I had to explain the filename deal each time. If the paragraph is advertising anything, it's OBK - a user who browses LT docs sees a mention of OBK and might even choose to use it, if for example ESPHome is too complicated and they want something more Tasmota-like.

Nobody called it "how to abandon obk". If someone wants to switch firmware, it's entirely their choice. I have nothing against "ESPHome to OBK" guides, if you ever decide to write one, as long as you keep it clean. 

My guide is purely informational, doesn't say a single bad word about OBK, and yet here you are calling it unethical. 

Feel free to put this conversion command on Elektroda. Heck, you could even provide UF2 files in GitHub builds, which would make migration even easier. I really don't care, and neither should you.