openshwprojects/OpenBK7231T_App #729: `SM2135_Current` doesn't work properly.

lmcd, 2023-03-14:
**Describe the bug**
SM2135_Current only seems to affect CW. The command for me has no effect on RGB at all.
For example, setting SM2135_Current 0 0, disables LED temperature altogether, but RGB is unaffected. Boosting this value has a noticeable impact on temperature brightness, but RGB remains unchained.

**Firmware:**
- Version: 1.15.565
- Device: EXTRASTAR E14 Smart Bulb
- Chip/model: BK7231T

**To Reproduce**
1. Set bulb to a temperature
2. Run `SM2135_Current` with a combination of values and note how the change is reflected as brightness
3. Set bulb to an RGB colour
4. Run `SM2135_Current` with a combination of values (including `0 0`) and note that nothing at all happens

**Additional context**
Flipping flag `9` on and off has no difference. I suspect this might be something to do with the communication protocol in `drv_sm2135`

openshwprojects, 2023-03-15:
Thank you for testing. Can you try with the latest update?
https://github.com/openshwprojects/OpenBK7231T_App/commit/8ce3c703c5743f5e3891da2719f9bf857f9c20fc
Again do your test, check if it works better now..?

lmcd, 2023-03-15:
Thanks for the quick release!
This partially-resolves the problem. The change now affects the light when showing an RGB colour, but it doesn't seem accurate. Setting `0 0` now makes the light quite dim, rather than turning it off completely.

openshwprojects, 2023-03-15:
Are you aware about the used current codes?
[image]

lmcd, 2023-03-15:
Yep, I've seen this.
Actually I think your code is correct, as I'm seeing 8 distinct steps for RGB as documented here: https://www.mikrocontroller.net/attachment/430042/SM2135E_zh-CN_en-US_translated.pdf

I think an explanation for why the stock Tuya bulb is still way brighter, is that maybe the original firmware is combining the RGB LEDs with the white ones to produce colours?

lmcd, 2023-03-15:
@openshwprojects if this turns out to be the case, can we have an option to tint all RGB colours with some added whiteness?

lmcd, 2023-03-15:
Found the problem: two identically packaged bulbs, with identical SKUs, markings etc have totally different chipsets and LED arrangements 😡

openshwprojects, 2023-03-17:
Well, at least now you can submit a teardown:
https://www.elektroda.com/rtvforum/topic3950844.html