openshwprojects/OpenBK7231T_App #730: How to add support for the DS18D20 temperature sensor with 1-Wire interface?

toneyche, 2023-03-15:
I have an MS-103 temperature control module from Moes, and I flashed it with OpenBK, which is great. Thank you for your work. However, I have an issue that I cannot drive the temperature probe. The probe is a DS18D20 that uses the 1-Wire protocol. How can I add this protocol? If possible, I'm willing to sponsor some fees to help solve this problem.
[image]
[image]

openshwprojects, 2023-03-19:
Hello, is this still active? I am planning to add this sensor.

toneyche, 2023-03-20:
Hello, thank you for your reply. I am currently using CHT8305 as a temporary replacement, but I still need to use DS18D20. Thank you. I have already paid $100 to express my gratitude for your work.

openshwprojects, 2023-03-20:
Thank you, your support is very helpful. It's also worth to see my Github profile, since several months we have at least one contribution each day.

I will look into that, and in a meantime, I must admit I haven't seen your device yet. Can you consider posting a quick teardown/review/pins description to our teardowns list?
https://openbekeniot.github.io/webapp/devicesList.html
so far we have nothing under DS18D20 keyword, so your device would bring new information.

toneyche, 2023-03-22:
Hi, I have posted a teardown of the device and its GPIO definitions. Thank you for your help, and I hope it will be useful for OpenBeken.

toneyche, 2023-04-18:
Have you made any progress?

openshwprojects, 2023-04-18:
Hello, I have several DS18D20  bought, but they are waiting for some time to be implemented. Yesterday I was improving connection speed for battery powered devices. Maybe in the upcoming days...

toneyche, 2023-04-19:
Thank you for your work and hope you finish it soon. I started another donation for you. Thank you.

Ginpachi98, 2023-04-21:
Looking forward to it too :D

awm1, 2023-05-31:
In the meantime which temperature sensor of similar form (waterproof casing) could I use? I am not able to find CHT8305 anywhere to be sold.

hojnikb, 2024-02-23:
Any news regarding support for this sensor?