openshwprojects/OpenBK7231T_App #732: Docker build succeeds, but outputs directory empty

lmcd, 2023-03-15:
I've succeeded in getting `OpenBK7231T_App` to build with Docker on an M1 Mac. A new directory `dev_20230315_195628` shows up in `output`, but it's empty.

lmcd, 2023-03-15:
I'll also add that dpkg was failing with some 404s - I resolved this by changing the `Dockerfile` with `--add-architecture arm64`

michaelkamprath, 2023-03-15:
I am pretty sure it'd be empty because you aren't using an x86 system and the cross compilers used are x86 executables. The cross compilers are probably simply not running because of the instruction set incompatibility, so it is unsurprising to me that you got so artifacts in the output directory.

I didn't test on a M1 Mac, but I do have access to one. I think the right way to make it work is to make sure the Mac know the Docker image is an x86 image, and thus trigger using the Rosetta emulator. I'll test things out and report back here.

lmcd, 2023-03-15:
@michaelkamprath - it does look like a full, comprehensive build is taking place. Takes a few minutes and reports back successfully.

Yes would be appreciated if you could have a quick test on M1.

michaelkamprath, 2023-03-24:
OK, so I got a chance to look into this. Its as I suspected. Without any special action, trying to build with the normal docker commands on a ARM Mac fails because the default ISA that docker on the ARM Mac builds is an ARM image, and the executables that are being used during the compile process are x86, and the docker build fails any (it starts off fine, but fails when the build process tries to install the `i386` platform). However, with some workaround, you can successfully get the software to build on a ARM Mac.

First, read [this article on how to build and run x86 docker images on an ARM Mac](https://blog.jaimyn.dev/how-to-build-multi-architecture-docker-images-on-an-m1-mac/). You need to use an experimental feature in docker called `buildx` to make that happen. Basically, the docker build command becomes:
```sh
docker buildx build --platform linux/amd64 -t openbk_build --build-arg USERNAME=$USER .
```
**IMPORTANT** - don't just run that command! Read the article I linked! There is some setup to your machine that you need to do.

Then, once the image build, you can build the OpenBK software with the `--platform` option too:
```sh
docker run --platform linux/amd64 -it -v "$(pwd)/..":/OpenBK7231T_App  openbk_build
```
Note as per the original docker instructions, both commands above should be run from the `docker` directory of this repository.

I will update the  Docker instructions soon.

PR #751 (2023-03-27): Added instructions for building with Docker on ARM Macs
Adds instructions on how to build the OpenBeken software on Apple Silicon Macs. This solves the issues that the compilers used by OpenBeken are built for x86 platforms.

closes #732
closed 2023-03-27 by openshwprojects