openshwprojects/OpenBK7231T_App #734: [Feature Request] Button role to support more than two channels.

ataa, 2023-03-16:
I have this 5 Channel power strip, It has single button, I want to be able to control all of 5 channels individually by multiple clicking on the button, at the moment the button role allows two channels only.

[image]

```json
"device_configuration": {
		"bt_type": 0,
		"ch_cddpid1": 9,
		"ch_cddpid2": 10,
		"ch_cddpid3": 11,
		"ch_cddpid4": 12,
		"ch_cddpid5": 13,
		"ch_dpid1": 1,
		"ch_dpid2": 2,
		"ch_dpid3": 3,
		"ch_dpid4": 4,
		"ch_dpid5": 5,
		"ch_num": 5,
		"crc": 100,
		"ffc_select": 0,
		"jv": "1.1.7",
		"module": "CB3S",
		"net_trig": 2,
		"netled1_lv": 0,
		"netled1_pin": 23,
		"netled_reuse": 1,
		"reset_t": 3,
		"rl1_lv": 1,
		"rl1_pin": 7,
		"rl2_lv": 1,
		"rl2_pin": 8,
		"rl3_lv": 1,
		"rl3_pin": 14,
		"rl4_lv": 1,
		"rl4_pin": 9,
		"rl5_lv": 1,
		"rl5_pin": 24,
		"total_bt_lv": 0,
		"total_bt_pin": 26,
		"total_stat": 2
	}
```

openshwprojects, 2023-03-16:
Why not use event handlers? The 5x click was not there, but I just added it:
https://github.com/openshwprojects/OpenBK7231T_App/commit/fd14d7d1f3711aa75f20d288611b48f1f005781d

ataa, 2023-03-16:
Thanks,
So I should write some script and assign it to the button or put it in autoexec.bat?
P.S. : The Auto build has some issues:
```json
make[1]: *** [Makefile:102: build-BK7231] Error 2
make: *** [Makefile:77: OpenBK7231N] Error 2
Error: Process completed with exit code 2.
```

ataa, 2023-03-16:
Flashed the latest release and created autoexec.bat with:
```
setChannelType 1 toggle
setChannelType 2 toggle
setChannelType 3 toggle
setChannelType 4 toggle
setChannelType 5 toggle
addEventHandler OnClick 26 ToggleChannel 1
addEventHandler OnDblClick 26 ToggleChannel 2
addEventHandler On3Click 26 ToggleChannel 3
addEventHandler On4Click 26 ToggleChannel 4
addEventHandler On5Click 26 ToggleChannel 5
```
Is this the right way to do it?

openshwprojects, 2023-03-16:
Hello, I think it might work...  altough... you don't need setChannelType 1 toggle, etc, etc, ToggleChannel will work for default type as well.

ataa, 2023-03-16:
After flashing the latest release, it does not boot anymore! not sure why!

openshwprojects, 2023-03-17:
What is the entire configuration of this device?

openshwprojects, 2023-03-17:
ToggleChannel [Ch] works for any Ch, even if it has no type 'toggle'.... I have both automatic test for that:
https://github.com/openshwprojects/OpenBK7231T_App/commit/d13743eb3810d5cb084408109552cc028c089c64
and I tested on my device, from a console level...
Futhermore, I know the ToggleChannel implementation.

Maybe there is some kind of bug  or issue somewhere else?

ataa, 2023-03-17:
So it might be a bug, if I remove `setChannelType 1 toggle` from `autoexec.bat`, press the button, the light for that channel turns ON for less than a second then turns OFF, but for some reason it does not turn on the relay.

```json
  "pins": {
    "7": "Rel;1",
    "8": "Rel;2",
    "9": "Rel;4",
    "14": "Rel;3",
    "23": "WifiLED;0",
    "24": "Rel;5",
    "26": "Btn;0"
  },
```

openshwprojects, 2023-03-17:
You are using Btn role which by default already toggles a channel on click and on double click. It's a configuration error.

If you want to avoid automatic button <->relay link, use Btn_ScriptOnly role.

ataa, 2023-03-17:
Yes, that was the reason. Thank you.

closed 2023-03-17 by ataa

openshwprojects, 2023-03-18:
Btw @ataa please consider posting your power strip docs (photos, etc, maybe from inside but not necessary, etc) to our templates list to help us:
https://openbekeniot.github.io/webapp/devicesList.html
You have very interesting devices. Thanks!

ataa, 2023-04-10:
Sure, I just received more devices to play with, I will document them all and post some information and photos to the forum.