openshwprojects/OpenBK7231T_App #735: RGBCW Bulb - Warm/Cold slider not working.

bmcsorley, 2023-03-16:
I have a Feit BR30/RGBW/CA/AG that I am working on teardown post for, but want to get the OpenBK App config working.

The bulb says RGBW, but there are dedicated Cold LED's and dedicated Warm LED's.  The LED driver is BP5778 (no CLK or DAT pins), and bulb has a CB2L module.  I have it configured to use 5 PWM channels right now.  If I toggle each in RAW PWM mode, I can turn on Cold White and Warm White.

But I'd like to use the LED RGB/CW/etc picker.  

If I don't have Flag 8 checked, both brightness and temperature slider control the brightness.  With flag 8 checked, the brightness slider works, but the temperature slider does nothing.

Thoughts?

openshwprojects, 2023-03-16:
If it's PWM RGBCW then you most likely have wrong channels order - it should be R G B C W, channel 1 is red (or 0 if you number from 0), channel 2 is green ,etc

bmcsorley, 2023-03-16:
Well now I feel silly.

I had RGB as channel 0,1,2.

But for CW and WW, I had channel 4 and 5.  Not sure why I skipped 3.

It is working now.