openshwprojects/OpenBK7231T_App #738: Lamp with additional RGB light - two brightness sliders?

ghoeffner, 2023-03-17:
Hello,

I recently acquired a Ledvance LED Panel Smart+ WiFi Planon Plus Backlight ceiling light.
It comes with a main WW/CW panel and an additional RGB-Light. How can I make both of them controlled independently? 
Currently whenever I choose an RGB color the CW is turned off and vice versa.

openshwprojects, 2023-03-18:
Hello, what kind of device do you have? Is it some kind of special LED lamp? Or do you want two brightness sliders for a common RGBCW device?
Please post a teardown, if you can:
https://www.elektroda.com/rtvforum/topic3950844.html
I must admit that I am not yet sure how to handle this case, I haven't yet seen any device requiring CW and RGB being on together. Futhermore, I even experienced the LED power supply turning off when I enabled all channels together, the overload protection was kicking in....
I can try to figure out something for you.
Do you just want a control from Home Assistant level, or does it have to work on our Web Panel as well?

ghoeffner, 2023-03-18:
Thanks for your response. I honestly did not try it with the tuya app, as I have no use for the lamp if I cannot use it without a cloud ;-) 

**Product advantages** 
- Easy and quick installation
- Main light and rear light
- Both lights are independent
- TW and RGB
[image]

There are other lamps such as a tuya powered outdoor floodlight that can run both RGB and whites without any issue from a power perspective. Check out "[SetOption37 128](https://tasmota.github.io/docs/Commands/#setoption37)" which I believe is what we need for this use case.

Would be awesome if you can help. I also plan on posting a teardown + template once I get it to work fully.

openshwprojects, 2023-03-18:
So basically what would you need? A separate dimmer slider and Enable button for CW, and a separate dimmer slider and enable button for RGB?

ghoeffner, 2023-03-18:
That is correct - a separate enable/disable and dimmer and they should not turn the other group off :-) 
Would be awesome if implemented and I am happy to test it!

openshwprojects, 2023-03-19:
Are you going to pair it with HA?

ghoeffner, 2023-03-19:
That would be awesome, but I could probably live without it :-)

openshwprojects, 2023-03-19:
I will think about it and provide you some kind of solution.

Regarding german version - we have a german forum, and a translator: elektroda.de so if you want to save some time, just post it there. Our team will do the translation to english.

ghoeffner, 2023-05-13:
Got it installed and paired with HA now. Would be awesome if there is an option to have two separate entities in HomeAssistant for the ambient and the main light.

bagges, 2025-11-11:
@ghoeffner Did you get the RGB stuff to work? Also, do you know if it's possible to flash the controller without desoldering it?