openshwprojects/OpenBK7231T_App #739: New web app tools/feedback thread

openshwprojects, 2023-03-18:
I am working on web app update, currently I have added a tools section. Tools sections has some options for users (energy metering calibration frontend), and some for devs (repeating event testing, DGR sending). 
[image]
If anyone here is reading, please play around with tools and tell me if they are working, is there anything we may add, etc.
But take CARE - if you, for example, try to calibrate your device, results will be saved in flash and you will lose old calibration!
PS: There is also tool to set RGBCW map

ekobres, 2023-03-20:
Send Tasmota Color Commands are failing for me:
```
Last command: DGR_SendRGBCW FEIT-LED-STRIP FF00000000
Last status: ERROR: Not enough arguments for this command (undefined)
```

Also, Ping-Pong animation for brightness and color for my LED strips appears to only do brightness, but when I hit "stop" it makes it non-responsive (very slow) - however, when I do this I see a lot of this in the log - which is suspicious:

```
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
Info:CFG:####### Flash Save LED #######
```

openshwprojects, 2023-03-21:
Thanks! You are correct, there is a documentation error/issue, with RGB send, documentation says it expects hex string, while hex string was not supported and it expected separate numbers... I have updated it to support both.

Hmmm saving led settings is a byproduct of the way that mechanism works, I will have to think how to solve it. Maybe I will add a dont-save flag to led control commands.

The Temperature and Dimmer animations are working on me on latest build.

DeDaMrAzR, 2023-04-01:
To confirm that calibration from web app works on BK7231T + BL0937 - tested on version 1.15.638