openshwprojects/OpenBK7231T_App #742: BK7231N CBLC9 flashed but not starting WiFi AP

electromaticboy, 2023-03-20:
Hi,

I have a few of these Feconn MR16 Smart Bulbs (https://www.amazon.co.uk/Compatible-SmartThings-2700K-6500K-Spotlight-FECONN/dp/B09K3GVLYT) that have a BK7231N board CBLC9 (https://www.elektroda.com/rtvforum/topic3951016.html)

I've flashed the latest firmware and I can see the bootloader CLI changes and has more commands but the WiFi AP doesn't come on automatically. I can launch it and connect to it manually or I can connect it to my router from the CLI and it gets an IP address but still the web server is not running and I don't know where to go from there.

The rest of the board is just a step-down converter and the LED drivers.

Is there anything I've missed? Why wouldn't it start the AP?

Also, if I restore the 2.00M backup, is it safe to assume that it will overwrite the RF part including the MAC address?

openshwprojects, 2023-03-20:
If the 2MB backup is from the same device, then overwriting RF/MAC will not change anything, because it will contain original MAC and correct RF.

Device connects to WiFi and responds to PING but web app is offline? are you sure? 

If your device connects to WiFi and gets IP, then how did you configure your device if WiFi AP doesn't come up?

Have you tried the "Erase all" tricks within our flasher?

I am not sure if I understand what your issue is.

electromaticboy, 2023-03-20:
Thanks for the quick reply!
> If the 2MB backup is from the same device, then overwriting RF/MAC will not change anything, because it will contain original MAC and correct RF.
Yes, it's from the same device before I flashed it with OBK
> Device connects to WiFi and responds to PING but web app is offline? are you sure?
I haven't run nmap to find all open services but I tried the IP address a couple of times from Firefox. Might have to try again because it keeps defaulting to https even when I type only http.
> If your device connects to WiFi and gets IP, then how did you configure your device if WiFi AP doesn't come up?
From the CLI, I issued the "softap wifiname wifipassword" command
> Have you tried the "Erase all" tricks within our flasher?
Yes, I've tried that and I did the "RF Restore Part". Tried twice doing the RF restore before flashing and after flashing, same result.
> I am not sure if I understand what your issue is.
I was just wondering if I was missing something obvious like setting autostart.bat or if there is a way to start the applications from the CLI. I was doing it after a long day so might have to try again but running "ps" didn't seem to show the web server running (guessing it should?).
> 1.15.611 ? Seems ok for me: [image]
I'll give it another try tonight.
Does it make any difference that the device was already paired to my WiFi? I can see the SSID and password in the .bin backup and I read that OBK will take those and connect to the WiFi automatically instead of starting in AP mode.

openshwprojects, 2023-03-20:
What is SoftAp command? Is it on our commands list? As far as I know, we don't have such command  yet.

Have you tried 5 quick power off/on cycles?

electromaticboy, 2023-03-20:
> What is SoftAp command? Is it on our commands list? As far as I know, we don't have such command yet.

I'll confirm the command tonight, I was very tired and can't remember exactly what I typed. I tried creating an AP by hand which worked, and I could connect to it (can't remember if that was with OBK or the original firmware though) and I also tried connecting it to my WiFi and it reported on the CLI the IP address encoded in HEX, confirmed it was the correct address with PING and tried the browser but nothing there (could be an issue with Firefox looking at the wrong port).

> Have you tried 5 quick power off/on cycles?

I tried but had the module desoldered from the PCB so couldn't double check that it was doing it properly. I didn't see the AP coming up and the CLI bootloader messages where all the same. Still, another thing to try tonight!

electromaticboy, 2023-03-20:
This is the bootloader log on USART1 after full erase, restore RF and flashing OpenBK7231N_QIO_1.15.620.bin

`bk_misc_init_start_type 0 0
prvHeapInit-start addr:0x414498, size:113512
[Flash]id:0xeb6015
sctrl_sta_ps_init
cset:0 0 0 0
bandgap_calm_in_efuse=0x64
[load]bandgap_calm=0x64->0x24,vddig=4->5
[FUNC]rwnxl_init
[bk]tx_txdesc_flush
[FUNC]intc_init
[FUNC]calibration_main
gpio_level=1,txpwr_state=0
                          user define rfcali mode:1
get rfcali_mode:1
calibration_main over
                     NO TXPWR_TAB_TAB found in flash
Load default txpwr for b:0xdab7c
Load default txpwr for g:0xdab8a
fit n20 table with dist:4
Load default txpwr for n40:0xdab98
Load default txpwr for ble:0xdba11
temp in flash is:350
xtal in flash is:12
xtal_cali:12
--init_xtal = 12
                idx:40
[FUNC]ps_init
[FUNC]func_init_extended OVER!!!

start_type:0
Version:
Initializing TCP/IP stack
app_init finished
ble mac:38-1f-8d-09-dd-b0
xvr_reg_init
h4tl_init-1 ok
hci_init ok
!!!!!!init_type=0
                 rwble_hl_init ok
rwble_init ok
rwip_driver_init ok
[ble_appm_send_gapm_reset_cmd]
tx_pwr_idx:16
enter dut mode
!!!!!!init_type=1
                 llm_init:312
[gapm_cmp_evt_handler] conidx:0,operation:0x1,status:0x0
cmd->addr.addr[5] :0
!!!!!!init_type=2
                 [gapm_cmp_evt_handler] conidx:0,operation:0x3,status:0x0
gapm_cmp_evt:GAPM_SET_DEV_CONFIG
gapm_cmp_evt:wait GAPM_GEN_RAND_NB
[gapm_cmp_evt_handler] conidx:0,operation:0x1a,status:0x0
gapm_cmp_evt:GAPM_GEN_RAND_NB
[gapm_cmp_evt_handler] conidx:0,operation:0x1a,status:0x0
gapm_cmp_evt:GAPM_GEN_RAND_NB
[gapm_cmp_evt_handler] conidx:0,operation:0x28,status:0x0
gapm_cmp_evt:BLE_STACK_OK
ate_start
temperature_type=2
`
Tried power cycling many times to no avail. Tried flashing the backup, same thing. None of them will autostart the AP, there is no web interface running on OBK even after connecting to the WiFi using "sta mywifiname mywifipassword" and confirming that the device is online.

electromaticboy, 2023-03-20:
Found the problem! I have somehow left behind a tiny spot of solder between CSN and GND when soldering TX1 and RX1. I couldn't even see it with a magnifying glass.