openshwprojects/OpenBK7231T_App #745: OTA update bricks device (BL602)

Switch123456789, 2023-03-22:
**Description:**

- My device immediately bricks whenever I apply any recent OTA update.

**Firmware:**
- Version: 1.15.622
- Device: Magic Home RGB-LED controller
- Chip/model: BL602

**To Reproduce**
1. Launch Web App & go to OTA
2. Select a image of a recent build & flash it
3. It will look like this and now the device is bricked
[image]

openshwprojects, 2023-03-22:
Sound's like a serious issue, are you able to determine when it started to happen?

Switch123456789, 2023-03-22:
I didn't update the firmware for like 2 month maybe. 2 days ago I tried to OTA-update with the latest build and the device died. Flashed it again the UART-way and it came back alive including my settings. It was the same thing with todays attempt, it died again.

Switch123456789, 2023-03-22:
I downgraded via OTA from the latest version to `1.15.361`, no issues! From there I tried this these updates:
OTA 500 OK
OTA 531 OK
OTA 551 OK
OTA 571 OK
OTA 581 OK
OTA 591 FAILED
Gonna try to find the exact one now.

openshwprojects, 2023-03-23:
Sadly I have issue with my BL602 dev board, not related to the OTA problem.

@Switch123456789 , thank you for analysis, this is very useful information.
Can you check if OTA works for https://github.com/openshwprojects/OpenBK7231T_App/releases/tag/1.15.623  ? 
I reduced binary size...

Switch123456789, 2023-03-23:
I OTA-updated from
`1.15.609` to
`1.15.623` and to
`1.15.624` no issues :)

openshwprojects, 2023-03-24:
So it's a binary size...

Switch123456789, 2023-03-31:
Does that mean the image is to big for the storage of the chip? (noob here)
BTW, I just flashed `1.15.637`, again no issues anymore. I'll keep a device wired for flashing, let me know if I can help testing and stuff.

Sidenote: The files this selector is referring to are not OTA-flashable, I am not sure if this is supposed to be like this.
[image]

closed 2023-03-31 by Switch123456789