openshwprojects/OpenBK7231T_App #749: Pin Settings for Tuya 5in1 RGBCCT LED Strip Controller

Matsched, 2023-03-26:
Hello guys!
ive just bought [this RGBCCT 5in1 LED Strip Controller Tuya](https://de.aliexpress.com/item/725319185.html) from Ali with a BK7231T and flashed openBeken on it. Im missing the controller in the Device Lust under /app? and have no clue which Pin is R,G,B,CW,WW,V etc.
Can i broke the device by try and error? Where can i get the information about my device?

thanks!

openshwprojects, 2023-03-26:
Hello, we are usually using forum for devices discussion:
https://www.elektroda.com/
but yes, you must open device (or read flash for analysis, before flashing obk) and see which GPIOs are used for PWMs. There is 6 possible PWMs in BK, so you can also resort to guessing.

Matsched, 2023-03-27:
thank you very much!

closed 2023-03-27 by Matsched