openshwprojects/OpenBK7231T_App #750: Inability to set MQTT QoS

chriskinal, 2023-03-26:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware:**
- Version 1.15.631
- Device Treatlife SS02
- Chip/model: BK7231T?
- Device config: All configurations

**To Reproduce**
Steps to reproduce the behavior:
1. Go to main menu on the device
2. Click on 'Config'
3. Click on 'Configure MQTT'
4. No filed to specify MQTT QoS.

On some networks it would be helpful to specify a higher QoS than the default setting of '0'. This would ensure subscribers such as Homeassistant stay in sync with the state of the device on less than ideal networks.

openshwprojects, 2023-03-27:
Hello, I can easily add  an access for that, maybe through a console command?

chriskinal, 2023-03-28:
Either a console command or add it to the MQTT settings page. My preference would be the settings page so all MWTT settings are in one place and be easily reviewed or changed.

openshwprojects, 2023-03-29:
Hmmm, from what I can see in the code, the publish QOS in our firmware seem to be 2.  Is QOS 2 a good setting for you, or do you just need it configurable?

chriskinal, 2023-03-30:
The two MQTT browsers I am using both show the QOS from the OBK devices as 0. I also checked with tcpdump and see the QOS is also 0. While QOS 2 certainly ensures 1 and only 1 copy of the message is received, it imposes some overhead. In a marginal network, this can slow the MQTT data transfers as the sender and receiver may end up waiting for "acks" that are lost.

I think having it configurable would allow one to try the different levels and measure the result. Then one can determine if the result meets their use case.