openshwprojects/OpenBK7231T_App #754: Error:MQTT:Publish err: ERR_MEM

ataa, 2023-03-28:
When I try to use 

Chipset: BL602 (RGBCW Bulb)
Flags: 10, 37
Version: 1.15.633 
1 drivers active (BP5758D), total 13

**To Reproduce**
1. Configure MQTT
2. Start HA discovery
3. Check the logs

**Additional context**
```json
Info:MQTT:Queued topic=pyrahass/light/officergbwbulb3_light/config, 1 items in queue
Info:MQTT:Publishing val (776 bytes) to pyrahass/light/officergbwbulb3_light/config retain=1
Error:MQTT:Publish err: ERR_MEM aka -1
```

ataa, 2023-03-28:
This is the first time that I tried to use MQTT and HA discovery on BL602. I don't have other BL602 based device to test.

ataa, 2023-04-09:
I am still getting `Error:MQTT:Publish err: ERR_MEM aka -1` on version 1.15.666

openshwprojects, 2023-04-10:
Hello, are you able to compile yourself? Maybe we just need to bump the buffer sizes here:
https://github.com/openshwprojects/OpenBL602/commit/60daa9fe7b2d216bf4550fb53bd894d4c21f82ff
Or.. I can try to increase it for you, but you'd need to test it.

ataa, 2023-04-10:
I already forked the repo but not sure how to change the buffer size and compile it for OTA.
Also noticed on BL602, the network hostname does not update if I change the name of the device.

openshwprojects, 2023-04-12:
Wait I decided to just increase buffer, can you try now?
https://github.com/openshwprojects/OpenBL602/commit/50ddebdf0088aa59b1fd531948f0a7c7f9f01d7f
https://github.com/openshwprojects/OpenBK7231T_App/commit/3ea87971954f8194b02b48fe11b2c907731b39d5

ataa, 2023-04-12:
Thanks, MQTT is working as expected now. The hostname issue is still there.

openshwprojects, 2023-04-12:
hmm you mean that host name is using default Bouffalo labs one?

I am aware about this but I haven't seen a place to change it so far, but if you have any idea, feel free to look for it into the SDK of BL602

ataa, 2023-04-12:
Did some research and couldn't find anything, seems to be set in their wifi blob but I might be wrong.

closed 2023-04-12 by ataa

ataa, 2024-02-01:
https://github.com/openshwprojects/OpenBL602/blob/e5769160cfc91a5fe36f040b3d5314e51eda3a28/components/bl602/bl602_wifidrv/bl60x_wifi_driver/wifi.c#L184