openshwprojects/OpenBK7231T_App #755: Weird current measurement BK7231T (WB2S) + HLW8012

kiklhorn, 2023-03-29:
LEDVANCE SMART+ SOCKET with 200W bulb plugged in
Chip HLW8012, driver BL0937
Power measurement is always OK (after calibration).
When SEL pin is set to "AlwaysLow" - current reading is OK (after calibration) 0 with zero power, ~0.9A with 200W
When SEL pin is set to "AlwaysHigh" - voltage reading is OK (after calibration)
then previous calibration untouched and
When SEL pin is set to "BL0937SEL" - voltage readings is OK, but current readings have big offset (reading fluctuating around ~1.5A with zero power, ~2A with 200W power)

kiklhorn, 2023-03-29:
I'm sure. VF1 and VF is OK (Otherwise power measurements "hopping") Flashed OpenBK7231T_UG_1.15.633.bin

openshwprojects, 2023-03-29:
First I was suspecting the PR with Power metering rewrite but if you have the same problem in 1.15.609 then that's not it... I must find my BL0937 device and double check.

kiklhorn, 2023-03-30:
I found where the problem is. The SEL pin is inverted (via transistor). According to photos from the Internet, I think this wiring type occurs quite often.
Is there any way to invert this in OpenBK?

Reflashed to ESPHome (LibreTuya project) and successfully tested:

ESPHome function code. Some little calibration is needed:
```
sensor:
  - platform: hlw8012
    model: HLW8012
    current_resistor: 0.001
    voltage_divider: 2351 
    sel_pin: 
      number: P26
      inverted: true
    cf_pin: P7
    cf1_pin: P8
    current:
      name: "HLW8012 Current"
    voltage:
      name: "HLW8012 Voltage"
    power:
      name: "HLW8012 Power"
    energy:
      name: "HLW8012 Energy"
    update_interval: 5s
    initial_mode: CURRENT
    change_mode_every: 8
```

openshwprojects, 2023-04-02:
Thank you, I will add an inverted SEL right away.

closed 2023-04-02 by openshwprojects

kiklhorn, 2023-04-07:
I tried version 1.15.661
I guessed wrong, the error is somewhere else. Maybe timing, throw away the first few samples?

openshwprojects, 2023-04-11:
So the error persists? Are the measurements wrong all the time?

DJBenson, 2023-04-11:
> I hope I found where the problem is, maybe. The SEL pin is inverted. I think this wiring type occurs quite often. Is there any way to invert this in OpenBK? Something as "BL0937SEL_n"
> 
> Reflashed to ESPHome (LibreTuya project) and successfully tested:
> 
> ESPHome working code. Some little calibration is needed:
> 
> ```
> sensor:
>   - platform: hlw8012
>     model: HLW8012
>     current_resistor: 0.001
>     voltage_divider: 2351 
>     sel_pin: 
>       number: P26
>       inverted: true
>     cf_pin: P7
>     cf1_pin: P8
>     current:
>       name: "HLW8012 Current"
>     voltage:
>       name: "HLW8012 Voltage"
>     power:
>       name: "HLW8012 Power"
>     energy:
>       name: "HLW8012 Energy"
>     update_interval: 5s
>     initial_mode: CURRENT
>     change_mode_every: 8
> ```

Slightly off topic question but is it possible to flash between OpenBeken and ESPHome (Libre-Tuya) using the rbl file generated by Libre-Tuya and back again?