openshwprojects/OpenBK7231T_App #756: Timer function

efal, 2023-03-31:
Is there a way to set up timer(s) like in tasmota, or are there any plans for it?

openshwprojects, 2023-03-31:
What kind of timer do you need? We have scripts with delays, repeating events, clock events...

efal, 2023-03-31:
Thank you. I have to look this up, but for user friendliness I'd prefer to set timers in the web interface, somehow like in tasmota as I already said. From my perspective timer functions are basic for IoT-Devices.

openshwprojects, 2023-03-31:
We can add timers in Web App, but how would you like them to work?

efal, 2023-03-31:
Don't get me wrong. You've done a great job allready.
Timer function should use settings for on/off commands on a daily, weekly basis with easy to adjust timings (dropdowns, input fields or similar). This could help less programming biased users to take advantage of openbeken. Thank you.

openshwprojects, 2023-04-02:
Hmm so basically you want a frontend for our existing features. I will think about it.

ilengyel, 2023-04-17:
Possibly a different enhancement request is Sunrise/Sunset support similar to https://tasmota.github.io/docs/Timers/#sunrise 

And indeed a nice to have is the webapp hosting a timer & settings script generator which could also incorporate: https://tasmota-tz.cloudfree.io/

MoritzThome, 2026-03-10:
It would be awesome to implement the tasmota timer/scheduling with the same api.
I often recommend or install openbeken/tasmote devices to non-nerd people as they don't need HA Server since they run standalone.
The Adnroid/IOS App `Tasmotrol` can control Tasmota and also OpenBeken Devices just fine, but timers are only working on tasmota. You can define timers from that easy app.