openshwprojects/OpenBK7231T_App #758: Miboxer FUT036w with CB3s Problems

MarcoL-81, 2023-04-01:
Hello, i have the led controller miboxer Fut036w with Wifi and 2,4gHZ.
i have flash with the ObenBK Firmware. 
The Flashing and the CRC is OK.
CRC matches 0x4ACF98A4!
Writing file data to chip successs.
When i connect the power i can see the Wifi. 
Then i can connect to my home Wifi
i can connect to mqtt.
But i cant control it the web.
i only can switch and dimm with the local switch, like before. 
and the flashing led on the pcb is also like before. it seems that the old firmware is active. but the  openbk is running with mqtt and so on.
what is my error?
[image]
[image]
[image]
[image]

openshwprojects, 2023-04-01:
Please post in our devices forum:
https://www.elektroda.com/rtvforum/forum390.html
but in general, this is a TuyaMCU device and may require further setup/research.

MarcoL-81, 2023-04-02:
ok, i make a post in the forum.
Zhe problem is that there is a second µC. I have dismount it and then its working.
But i will make some adjustments about the BTN_SmartLED 
https://www.elektroda.com/rtvforum/topic3969126.html

closed 2023-04-06 by openshwprojects