openshwprojects/OpenBK7231T_App #759: Self-Signed MQTT Server Connection Fails

skellycode, 2023-04-02:
**Describe the bug**
Connections to Secure MQTT server (i.e. port 8883) fail for self-signed certificates

**Firmware:**
- Version: 1.15.642
- Chip/model: BK7231T

**To Reproduce**
Steps to reproduce the behavior:
1. Go to MQTT settings and add the credentials for an MQTTS server that is running a self-signed certificate
2. See connection error on main summary page ('disconnected')

**Recommendations**
- Perhaps adding an 'ignore security errors' flag in the MQTT settings as a check-box.  This would allow users to choose whether they wish to accept the security errors associate with self-signed certificates.  Users who do not use self-signed certificates would be able to leave the box unchecked.

skellycode, 2023-06-04:
Any update on this @openshwprojects?
This is breaking for me as I cannot use my OpenBK devices without an MQTTS connection...

openshwprojects, 2023-06-04:
Hello @skellycode , I apologize for the delay, I keep in mind this feature request, but I have already a backlog of requested in queue waiting to  be processed, not to mention some requests from the direct contributors/donators (for example, a guy sent me a PIR sensor....).
Do you have any C knowledge? Maybe we could look into that MQTTS together, that could be faster.

So you basically need a MQTT secure connection? So we would need to enable TLS support in the SDK?

Our MQTT library resides in LWIP, it is both present in BK7231T and BK7231N SDKs.

skellycode, 2023-06-05:
thanks for the reply - unfortunately I do not know any C (I mainly write R/Python).  I tried looking at the project for LWIP but I don't know what it is and got lost pretty quickly 😆 

> So you basically need a MQTT secure connection? So we would need to enable TLS support in the SDK?

Yes that's right - it would be a case of enabling secure connections for MQTT - AND allowing 'insecure' connections (i.e. self-signed certificates)