openshwprojects/OpenBK7231T_App #761: Variable does not expand in script

hftsai256, 2023-04-02:
**Describe the bug**
Script variables described in `docs/constants.md` would not expand in most scripting commands.

**Firmware:**
- Version 1.15.623
- Device? DayBetter 800 Lumens
- Chip/model: BK7231N
- Device config? Flag 18/27/37 enabled

**To Reproduce**
The outcome of the following script is not expected:
```
init:
  BP5758D_Map 2 1 0 3 4
  led_enableAll 1
  led_dimmer 100
  led_temperature 312
  BP5758D_Current 36
  goto mqtt_poll

mqtt_poll:
  if $MQTTOn==1 then goto publish
  delay_s 1
  goto mqtt_poll

publish:
  publish led_enableAll $led_enableAll
  publish led_dimmer $led_dimmer
  publish led_temperature $led_temperature
  goto post_init

post_init:
  echo "********************************************"
  echo $MQTTOn
  echo $led_enableAll
  echo $led_dimmer
  echo $led_temperature
```
The log shows:
```
Info:MQTT:Publishing val $led_enableAll to obk177FF51C/led_enableAll/get retain=0
Info:MQTT:Publishing val $led_dimmer to obk177FF51C/led_dimmer/get retain=0
Info:MQTT:Publishing val $led_temperature to obk177FF51C/led_temperature/get retain=0
Info:CMD:"********************************************"
Info:CMD:1
Info:CMD:1
Info:CMD:100
Info:CMD:312
```
And thus Home Assistant is receiving literal values such as `$led_enableAll`, not the actual values I would like to report from the device on boot.

The original intention was to follow the discussion from #623 and to have the device report its configured states on boot.

Also, the device would not push to the topic `obk1780C465/led_basecolor_rgb/get` when setting RGB colors. That's probably a separate bug.

openshwprojects, 2023-04-02:
Hello, the quickest solution is to use publishInt. The publish is not expanding yet because of a simply missing expand mechanism for strings, however, I think I can also fix it today for you, so both approaches will work.

But in general - a 5 seconds fix is to just replace "publish" with "publishInt".

hftsai256, 2023-04-02:
I was expecting similar behavior as bash variable expansion. However, if this is intended/expected then I agree to move forward with `publishInt` route rather than changing how `publish` behaves.

openshwprojects, 2023-04-02:
I also submitted fix your for initial issue:
https://github.com/openshwprojects/OpenBK7231T_App/commit/7714a0ab25b908a40e7410705c4d86fc27dcfa63
Please check and let me know if it works., but in general, if new issue arise, please open new ticket, so I can see it, or message me on Elektroda.

hftsai256, 2023-04-02:
Had it run on one of the devices and it can publish 3 out of 4 LED properties on boot:
[x] `led_enableAll`
[x] `led_temperature`
[x] `led_dimmer`
[ ] `led_basecolor_rgb`

Actually the last one never gets published unless I issue the manual command `publish led_basecolor_rgb xxxxxx` through webapp interface.
```
Info:MAIN:HA discovery is scheduled, but MQTT connection is not present yet
Info:MQTT:mqtt_connection_cb: Successfully connected
Info:MQTT:mqtt_subscribed to obk177FF51C/+/set
Info:MQTT:mqtt_subscribed to bekens/+/set
Info:MQTT:mqtt_subscribed to cmnd/obk177FF51C/+
Info:MQTT:mqtt_subscribed to cmnd/bekens/+
Info:MQTT:mqtt_subscribed to obk177FF51C/+/get
Info:MAIN:Will do request HA discovery now.
Info:MQTT:Queued topic=homeassistant/light/OpenBK7231N_177FF51C_light/config, 1 items in queue
Info:MQTT:Publishing val (828 bytes) to homeassistant/light/OpenBK7231N_177FF51C_light/config retain=1
Info:MQTT:Publishing val 1 to obk177FF51C/led_enableAll/get retain=0
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic obk177FF51C/led_enableAll/get
Info:MQTT:MQTT in topic obk177FF51C/led_enableAll/get
Info:MQTT:MQTT topic not handled: obk177FF51C/led_enableAll/get
Info:MQTT:channelSet NOT 'set'
Info:MQTT:Publishing val 312 to obk177FF51C/led_temperature/get retain=0
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic obk177FF51C/led_temperature/get
Info:MQTT:MQTT in topic obk177FF51C/led_temperature/get
Info:MQTT:MQTT topic not handled: obk177FF51C/led_temperature/get
Info:MQTT:channelSet NOT 'set'
Info:MQTT:Publishing val 100 to obk177FF51C/led_dimmer/get retain=0
Info:MAIN:Time 15, idle 164671/s, free 71304, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38 
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic obk177FF51C/led_dimmer/get
Info:MQTT:MQTT in topic obk177FF51C/led_dimmer/get
Info:MQTT:MQTT topic not handled: obk177FF51C/led_dimmer/get
```

openshwprojects, 2023-04-02:
Actually this is expected. The code works in such a way, that it checks in which mode is light currently, and if it's in RGB, it publishes RGB, and if it's in CT, it publishes CT.  Do you need a different behaviour for your purpose?

openshwprojects, 2023-04-02:
Related code snippets:
[image]
[image]

hftsai256, 2023-04-02:
It doesn't quite resemble what I observe on those devices, however. I tried publishing `ffff00` on `cmnd/obk177FF51C/led_basecolor_rgb` and the light turned yellow (as expected), but did not reply on topic `obk177FF51C/led_basecolor_rgb/get` which HASS is listening on. It is also missing from periodic self-state broadcasting (flag 2) as well when operating on RGB mode:
```
Info:MQTT:Publishing val 4 to obk177FF51C/sockets retain=0
Info:MQTT:Publishing val -33 to obk177FF51C/rssi retain=0
Info:MQTT:Publishing val 55 to obk177FF51C/uptime retain=0
Info:MQTT:Publishing val 71592 to obk177FF51C/freeheap retain=0
Info:MQTT:Publishing val 10.0.0.121 to obk177FF51C/ip retain=0
Info:MQTT:Publishing val 1 to obk177FF51C/led_enableAll/get retain=0
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic obk177FF51C/led_enableAll/get
Info:MQTT:MQTT in topic obk177FF51C/led_enableAll/get
Info:MQTT:MQTT topic not handled: obk177FF51C/led_enableAll/get
Info:MQTT:channelSet NOT 'set'
Info:MQTT:Publishing val 100 to obk177FF51C/led_dimmer/get retain=0
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic obk177FF51C/led_dimmer/get
Info:MQTT:MQTT in topic obk177FF51C/led_dimmer/get
Info:MQTT:MQTT topic not handled: obk177FF51C/led_dimmer/get
Info:MQTT:channelSet NOT 'set'
```
Is it possible that `shouldSendRGB()` in your code above could return `0` in my case?

openshwprojects, 2023-04-02:
Nice, you are correct. I have pushed a fix:
https://github.com/openshwprojects/OpenBK7231T_App/commit/e374b83d21489e58ae87d187c8075fb9dc54ad16
In the past we didn't have automatic RGBCW controls for I2C led drivers, that's why everything is checking not for the driver, but for "forceRGBCW flag". The issues are caused by partial migration from that standard.
Can you check now if it's working better for you?

hftsai256, 2023-04-03:
This build works great. Thanks for your help.

closed 2023-04-03 by hftsai256