openshwprojects/OpenBK7231T_App #762: [BL602] Crashes with Flag12 turned on

arhi3a, 2023-04-02:
**Describe the bug**
MagicHome device crashes after few seconds with _Flag 12 - [LED] Remember LED driver state (RGBCW, enable, brightness, temperature) after reboot_ turned on.
One particular danger I find with this bug is that after the crash device is setting all RGB leds to 100% brightness no matter the previous state of Toggle.

Last part of log with device connected to uart
```
[lwip] netif status callback
  IP: xxx.xxx.xxx.xxx (manual_edit)
  MK: 255.255.255.0
  GW: xxx.xxx.xxx.xxx (manual_edit)
[WF][SM] Exiting wifiConnected_ipObtaining state
[WF][SM] IP GOT IP:xxx.xxx.xxx.xxx (manual_edit), MASK: 255.255.255.0, Gateway: xxx.xxx.xxx.xxx (manual_edit), dns1: xxx.xxx.xxx.xxx (manual_edit), dns2: xxx.xxx.xxx.xxx (manual_edit)
[WF][SM] State Action ###wifiConnected_ipObtaining### --->>> ###wifiConnected_IPOK###
[WF][SM] Entering wifiConnected_IPOK state
[APP] [EVT] GOT IP 12504
[SYS] Memory left is 104080 Bytes
[APP] [EVT] GOT IP 12504
[SYS] Memory left is 104080 Bytes
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED - 4
-----------------> AABA Request:
    A-MSDU: Permitted
    Block Ack Policy: Immediate Block Ack
    TID: 0
    Number of Buffers: 64
-----------------> AABA Response:
    A-MSDU: Not Permitted
    Block Ack Policy: Immediate Block Ack
    TID: 0
    Number of Buffers: 8
Info:MAIN:Time 12, idle 0/s, free 104080, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 2/21 
Current task sp data:
RA:30303030, mstatus:80007880
A0:4202a5a8 A1:00000000 A2:00000000 A3:00000000 A4:00000000 A5:00000000 A6:00000000 A7:00000000
T0:98000000 T1:000007ff T2:00000000 T3:00000000 T4:00000000 T5:00000000 T6:33333200
S0:30303030 S1:30303030 S2:30303030 S3:30303030 S4:30303030 S5:30303030 S6:30303030 S7:30303030
S8:30303030 S9:30303030 S10:30303030 S11:30303030
Exception Entry--->>>
mcause 38000002, mepc 30303030, mtval 00000000
Exception code: 2
  msg: Illegal instruction
```

**Firmware:**
- Version 1.15.639 (reproduced also in 1.15.645)
- Device? MagicHome
- Chip/model: BL602
- Device config?
Default connected to WiFi
with assigned pins:
  "pins": {
    "3": "PWM;1",
    "4": "PWM;0",
    "20": "PWM;3",
    "21": "PWM;2"
  },

**To Reproduce**
Steps to reproduce the behavior:
1. set _Flag 12 - [LED] Remember LED driver state (RGBCW, enable, brightness, temperature) after reboot_ to ON
2. Reboot device 
3. Wait few seconds for device to crash.

**Fixing crash loop**
1. Reboot device
2. Spam http api request as device is booting (There is a brief moment where device is connected to Wi-Fi but not crashed yet) http://<device_ip>/cfg_generic?flag12=0&setFlags=1
3. Reboot device

cirelkc, 2023-10-19:
second this, any timeline to fix this please?

cirelkc, 2023-10-19:
also setting the start values to -1 doesnt work