openshwprojects/OpenBK7231T_App #763: RGBCW Bulbs with no COOL LEDs issue. Inaccurate Temperature setting

panchomira, 2023-04-03:
_First at all. Thanks to the owner of this project! Its amazing to be able to really be the owners of the products that we buy, ant the possibility to run open source firmware._  

[image]

**LAMP:** Etheos 9w E27
**Chip: Chipset:** BK7231N
**Firmware:** Version: 1.15.637

**The problem that, besides me, I saw another one have it and probably a few more will have.** 

As far as I can tell, my bulb does not have a cool white LED. Therefore, the original Tuya firmware achieves Cool white with RGB LEDs.

[image]

Currently, I am using Flag 24 ([LED] Emulate Cool White with RGB in device with four PWMS - Red is 0, Green 1, Blue 2, and Warm is 4), which tries to do the same thing, but it's way off because it doesn't work the same way as the original firmware.

**In the original firmware, when you set the temperature to around 4500K, all the warm white LEDs and RGB LEDs turn on (although I don't think they're all at the same intensity). Then, when you select the coolest white, only the warm white LED, the blue and green RGB LED turn on, not red.**
_See documentation at the end_

I tested the white balance on the original firmware using a camera, and it's pretty accurate in terms of the Kelvin temperature. Even when emulating temperature with RGB, it produces very little color in the final light projection, staying very true to a pure white.

**Now, on OpenBK, it goes with all the RGB LEDs as soon as soon as the lamp its not in the warmest setting, and stays the same (but changing the intensity of all RGB LEDs) even in the Coolest setting (not the way the original firmware does it)** 
_See documentation below_

Here are screenshot of the configuration and how it reflects on the LEDs of the bulb. Both in stock firmware and OpenBK
Original Firmware: 

Warmest
[image]
[image]
Middle
[image]
[image]
Coolest
[image]
[image]

OPENBK firmware:

Warmest: (So far so good)
[image]
[image]
Middle: (Not ok)
![Captura de Pantalla 2023-04-03 a la(s) 04 02 48](https://user-images.githubusercontent.com/98633070/229439729-b3397408-22
[image]
82-4383-8e67-d2c039e8ebf5.jpg)
Coolest: (the worst) here the white LEDs are turned off, which in the original firmware doesn't happen.
[image]
[image]

I would love to solve this issue because the temperature is not usable in the current state of OpenBk and this type of lamp. However, I have little to no knowledge in writing firmware.

If someone can help me with this, I am willing to spend some hours trying to calibrate it.

openshwprojects, 2023-04-03:
Hello, do you know the math behind it?

I have no experience when it comes to colours, but I know where the related code snippet is.
I have moved it to separate function:
https://github.com/openshwprojects/OpenBK7231T_App/commit/3ff88ff989c260720a07133d2aec01f3fd34eff6

With Github you can do a fork and you will get automatic github build on your fork and in potential pull request.

Are you interested in changing this function to reflect the desired behaviour?
[image]

The +0 is Red, the +1 is Green, and +2 is Blue.

If I understand correctly, you just want to add a fractions there? Like chVal * 0.5 to make red contribution smaller? Etc etc?

openshwprojects, 2023-04-03:
Wait, is it a PWM based device or a I2C driver? Because the I2C colour setting does not currently go through the LED_ApplyEmulatedCool route.

panchomira, 2023-04-03:
> Wait, is it a PWM based device or a I2C driver? Because the I2C colour setting does not currently go through the LED_ApplyEmulatedCool route.

Hello! Thanks for the super fast response! 

I forgot to mention that my bulb is using the BP5758D Driver. 
What I just tried (but didn't work) was to do a calibration with "led_gammaCtrl" to make the "white" RGB perform as the coolest setting in white mode (because the coolest mode tries to use all RGB like if it was white). I managed to do it, but it doesn't apply to the "coolest white" with RGB emulation. 

Im totally willing to learn and fix it. (as far as I can haha)

openshwprojects, 2023-04-03:
I have corrected the code so the calculation is shared between I2C drivers and PWM.

https://github.com/openshwprojects/OpenBK7231T_App/commit/6b502cfbb78c41320ddd09162b15d9d114118439

This function:
[image]
maps cool value to RGB.

You can edit it freely, if you think that we should use different fractions for each colour.

Maybe we should use luminance algorithm?
[image]

openshwprojects, 2023-04-03:
This approach allows you to change the fractions of R G B for cool emulation. But... do you also want to keep Warm LEDs on with the coolest option?

panchomira, 2023-04-03:
> This approach allows you to change the fractions of R G B for cool emulation. But... do you also want to keep Warm LEDs on with the coolest option? 

Amazing! And thanks! 

About the Warm LEDs, yes, we should keep it on since is the way the original firmware achieves color accuracy and brightness. As I mentioned on the post, once the setting is in its coolest, only the blue and green are turned on (besides the Warm LEDs, of course), and in the middle, the three RGB are turned on. I guess this has to do with color accury. If I find a way to emulate this maybe we can find a fix for all the bulbs that don't have the Cool LEDs from now on.

So, I'll take a look at the code right not and start thinking about it. I have to sleep in a few mins but I'll keep texting here for future considerations on how to fix it.

openshwprojects, 2023-04-07:
You should only modify finalRGBCW inside LED_I2CDriver_WriteRGBCW to suit your needs, no need for PWM calls.

ryan-ronnander, 2026-09-28:
For anyone picking this up later: the stock Tuya firmware has a `mix4to5: 1` setting that does its cool white emulation with a fixed table.

This mix table has been reconstructed from the firmware dump from one of my Tuya-based SYLVANIA smart bulbs.

Position 0 is the warmest end and 100 the coolest, and values between rows are interpolated linearly:

| Position % | Red | Green | Blue | Warm white |
| --- | --- | --- | --- | --- |
| 0 | 0 | 0 | 0 | 100 |
| 11 | 0 | 3 | 1 | 99 |
| 22 | 0 | 16 | 8 | 93 |
| 33 | 0 | 30 | 14 | 86 |
| 44 | 0 | 40 | 20 | 81 |
| 55 | 0 | 53 | 26 | 75 |
| 66 | 0 | 65 | 33 | 69 |
| 77 | 0 | 76 | 38 | 64 |
| 88 | 0 | 86 | 43 | 59 |
| 100 | 0 | 100 | 50 | 53 |

Red is never used, and the warm white LED never drops below about half.

Flag 24 instead copies the cool value into all three RGB channels, saturating them by about 4000 K, and fades the warm white to zero at the cool end, which matches the white LEDs going dark reported here. The two curves plotted against each other, from [this Elektroda post](https://www.elektroda.com/rtvforum/topic4034705.html#21974467):

[image]

The table comes from this particular bulb's firmware. I've gone ahead and scanned a bunch of other stock Tuya bulb firmwares that do the same mix (some turn it on with `mix4to5: 1` like mine, other firmware builds with `mixway: 3` and a `tempmix` table), and each one carries its own table, tuned for that product:

[image]

I know some things have improved from this issue, but there's not much you can do in OpenBeken to get a "perfectly stock" Tuya color experience right now with bulbs like this. I'm working on compiling a new GitHub repo to post a bunch of info around this and these types of smart bulbs so projects like ESPHome and OpenBeken can benefit.