openshwprojects/OpenBK7231T_App #765: RESTful `api/pins`: `deviceCommand` updates startup command

hftsai256, 2023-04-03:
**Describe the bug**
At `rest_interface.c:945`, `deviceCommand` on `api/pins` will invoke `CFG_SetShortStartupCommand_AndExecuteNow()`, which would also update the startup command. I'm wondering if this is intentional, because (at least in my application) one would have to invoke another HTTP GET over `startup_command` to restore the original config.

**Firmware:**
- Version 1.15.646
- Device? Probably irrelevant
- Chip/model: BK7231N
- Device config? Probably irrelevant

hftsai256, 2023-04-03:
I think that's fine for now as there are easy workarounds. I'm bringing this up for an open discussion to see if it is safe for everyone to make this API more intuitive.

iprak, 2023-04-08:
Sorry I do not remember working in that area, the change probably was made 8-9 month ago. I did format the document though.