openshwprojects/OpenBK7231T_App #766: BK7231N not connecting to wifi when CCMP 256 or GCMP 256 ciphers are enabled on router

00svd00, 2023-04-03:
Router model: Mikrotik hap ac3 (with wifiwave2)
Router wifi security config:
  authentification types: WPA2 PSK; WPA3 PSK
  ciphers: CCMP; CCMP 256; GCMP; GCMP 256

  
**Firmware:**
- Version 1.15.648
- Device? WiFi DIY Switch (Generic)
- Chip/model: BK7231N
- Device config? Default/Not matters

**To Reproduce**
If CCMP 256 or GCMP 256 are allowed on router BK7231N will be constantly failing to connect
On router it looks like this:
 apr/04 02:47:56 wireless,info CC:8C:BF:C3:5A:CB@cap-wifi1 connected, signal strength -46
 apr/04 02:47:57 wireless,info CC:8C:BF:C3:5A:CB@cap-wifi1 disconnected, connection lost, signal strength -46
 apr/04 02:47:57 wireless,info CC:8C:BF:90:0D:77@wifi_2ghz connected, signal strength -68
 apr/04 02:47:57 wireless,info CC:8C:BF:90:0D:77@wifi_2ghz disconnected, connection lost, signal strength -68
 apr/04 02:48:09 wireless,info CC:8C:BF:C3:5A:CB@cap-wifi1 connected, signal strength -46
 apr/04 02:48:09 wireless,info CC:8C:BF:C3:5A:CB@cap-wifi1 disconnected, connection lost, signal strength -46
 apr/04 02:48:09 wireless,info CC:8C:BF:90:0D:77@wifi_2ghz connected, signal strength -69
 apr/04 02:48:09 wireless,info CC:8C:BF:90:0D:77@wifi_2ghz disconnected, connection lost, signal strength -69
 apr/04 02:48:16 wireless,info F4:CF:A2:98:E6:E0@wifi_2ghz connected, signal strength -45

**Additional context**
For me only BK controllers are affected. All other wifi clients (including ESP controllers) are working with any ciphers allowed

openshwprojects, 2023-04-06:
Hello. Does it also happen on T?

00svd00, 2023-04-13:
> Hello. Does it also happen on T?

Hello! I don't know. I only have these relays for now
https://www.elektroda.com/rtvforum/topic3895572.html#20033093
New revisions are based on BK7231N and I don't own old revisions