openshwprojects/OpenBK7231T_App #767: An error occurred while following the instructions to create a Docker build.

toneyche, 2023-04-05:
I am using Ubuntu 22.04 to build firmware with Docker. The first step was completed successfully with the command "docker build -t openbk_build --build-arg USERNAME=$USER .". However, I encountered an error when executing the second step. The error message is as follows:

root@DockerHost:/home/hatel/OpenBK7231T_App/docker# docker run -it -v "$(pwd)/..":/OpenBK7231T_App  openbk_build
\\/build_target_platforms.sh: line 3: shell: command not found
****************************************
****************************************

Building OpenBeken for the following platforms: OpenBK7231T OpenBK7231N OpenXR809 OpenBL602 OpenW800 OpenW600
Building OpenBeken with version "dev_20230405_204440"

****************************************
Building platform: OpenBK7231T
****************************************

make -C sdk/OpenBK7231T/platforms/bk7231t/bk7231t_os APP_BIN_NAME=OpenBK7231T_App USER_SW_VER=dev_20230405_204440 clean
make[1]: *** sdk/OpenBK7231T/platforms/bk7231t/bk7231t_os: No such file or directory.  Stop.
make: *** [Makefile:140: clean] Error 2
make APP_NAME=OpenBK7231T TARGET_PLATFORM=bk7231t SDK_PATH=sdk/OpenBK7231T APPS_BUILD_PATH=../bk7231t_os build-BK7231
make[1]: Entering directory '/OpenBK7231T_App'
git submodule update --init --recursive --remote
error: could not lock config file .git/config: Permission denied
error: could not lock config file .git/config: Permission denied
fatal: Failed to register url for submodule path 'sdk/OpenBK7231N'
make[1]: *** [Makefile:34: submodules] Error 128
make[1]: Leaving directory '/OpenBK7231T_App'
make: *** [Makefile:74: OpenBK7231T] Error 2
................

openshwprojects, 2023-04-06:
Hmm I haven't used docker build, it is very easy to use online github builds, they are working also in forks and in the pull requests.
But maybe the Docker support contributor can shed some light on to issue - @michaelkamprath  can you help?

michaelkamprath, 2023-04-06:
It looks like your docker container didn't build successfully, notably it failed to build the container's linux image fully (missing certain commands such as `shell`). Please inspect the output of the build command more closely. Also ensure that you docker install is correct (e.g., it has proper permissions), and possibly repeal the base image for this build environment (`docker pull ubuntu:22.04`).

michaelkamprath, 2023-04-06:
Oh wait, I can confirm that the build is breaking now in the way. That means something changed with the base image. Investigating ...

openshwprojects, 2023-04-06:
There is a not yet fixed BL602 function call little issue, maybe that's it? But it would break BL602 only...

michaelkamprath, 2023-04-06:
It would appear the problem is in the Makefile, not the docker build. For what ever reason the `git submodule update` isn't working in pulling the needed sdk code. I've confirmed this is true outside of the docker environment too. I don't have time to solve it right now ... leaving this note here incase someone else can spot the issue ...

openshwprojects, 2023-04-06:
For which SDK?

toneyche, 2023-04-07:
Yes, I have built it online, thank you.

michaelkamprath, 2023-05-07:
I believe this issue is the same one as identified in #809.