openshwprojects/OpenBK7231T_App #768: TasmoAdmin runtime stats are way off

Wolfieeewolf, 2023-04-06:
**Describe the bug**
When using TasmoAdmin runtime stats they are way off. See attached image

**Firmware:**
- Version 1.15.656
- Device? All devices
- Chip/model: BK7231T/BK7231N
- Device config? Yes

[image]

openshwprojects, 2023-04-06:
Maybe it needs NTP running? Or is it uptime in seconds?

What do you get if you use http://192.168.0.202/cm?cmnd=status (enter your IP inside of mine)?

I get: "Uptime":"0T12:44:04","UptimeSec":45844 in that JSON, which seems ok ?

Wolfieeewolf, 2023-04-06:
I get this with the /cm?cmnd=status 

[image]

When I run the NTP driver it seems to have fixed the problem.

openshwprojects, 2023-04-07:
Sorry but are those valid values? Are they invalid without NTP? NTP should not be required for uptime counting..

Wolfieeewolf, 2023-04-10:
Just did some testing with the uptimes with and without NTP.

With - Uptime : "0T00:00:55" - UpTimeSec : 55 - Matches uptime in TasmoAdmin

[image]

Without - Uptime : "0T00:01:14" - UpTimeSec : 74 - Doesn't match uptime in TasmoAdmin

An interesting note is that the StatusTIM times are all wrong 

[image]

This would explain the huge Uptime number in TasmoAdmin if it is calculating the uptime from 1970.

openshwprojects, 2023-04-10:
They are 1970 while you don't have NTP running.

Wolfieeewolf, 2023-04-10:
Correct. Seems to work fine when the NTP is running

Dheenhasty, 2023-04-11:
Yes the Tasmota schema use the start time to do his calculation ..... so NTP is needed. i saw that in tasmoadmin too

Wolfieeewolf, 2023-08-18:
Closing this as the NTP driver has fixed the issue

closed 2023-08-18 by Wolfieeewolf