openshwprojects/OpenBK7231T_App #774: Make PWM frequency configurable

zolg, 2023-04-10:
For now (at least BK7231x devices) have two options for PWM frequency:
1000Hz and 600Hz (activated when Flag 11 set)

However even "fast" 1kHz PWM when used on light is not as fast as it should be. Being invisible in static scenes it provide strong stroboscopic artifacts on moving objects.
For example in the light of PWM driven cheap smart lamps flashed with OpenBeken firmware shower water streams breaks in to series of flickering individual water drops. 
Same shower stream in the light of the **same lamps** with original Tuya firmware looks normal without any artifacts.

Measurement shows that **original firmware use much higher PWM frequency**: some use 5kHz, and some even 10.

Custom built OpenBeken whith 
```
#define PWM_FREQUENCY_SLOW 5000 //Slow frequency for LED Drivers requiring slower PWM Freq
#define PWM_FREQUENCY_DEFAULT 10000 //Default Frequency
```
works like a charm on corresponding lamps providing artifacts-free lighting.

It would be nice to **have PWM frequency configurable** in official build. 
There are still some free free bytes in mainConfig_t struct and it would need just a dozen lines of code to store pwmfrequency in it. But I'm unfamiliar with OBk config system to make PR _non-breaking compatibility with existing config_ struct, stored in flash

So could you please implement storing pwmfrequency in config? Or tip me how to store additional variable in config _without breaking compatibility_ and I'll make a PR.

openshwprojects, 2023-04-10:
Hello, how many bytes do you need to store that frequency?

Basically you need to take 4 bytes or so from config (maybe 2) and then in config load do if(those_bytes == 0) then set default value.

And that should be all needed to get it running correctly... because unused bytes in config are zeroed, I think.

Then you can just simply include the config header from HAL files on platforms and use given value...

Anyway, your idea is very good and it was on my TODO list for very long time, but it's nice to see someone wanting to help. Feel free to create a pull request.

zolg, 2023-04-10:
First of all thank you for your fast reply

> Hello, how many bytes do you need to store that frequency?

_pwmfrequency_ in hal_pins_bk7231.c  is int, so most straight way to store it in config is to store (4 byte) int. Also we could probably save couple of bytes and store (2 byte) ushort (unlikely that PWM frequency above 65kHz would be needed or even possible). But I've not looked in to code and specs deeply and not sure if config struct is not 32 bit aligned.
On one hand 2 bytes does not cost that much in tooday's world, on another - sizeof(mainConfig_t) is quite close to 2k limit now.

> Basically you need to take 4 bytes or so from config (maybe 2) and then in config load do if(those_bytes == 0) then set default value.

Please correct me if I'm wrong:
Adding variable to the end of _mainConfig_t_ (after _initCommandLine_) would change size of mainConfig_t. And therefore the actual block of data CFG_CalcChecksum calculated against.
As checksum is not "sum" but some sort of CRC addition of new bytes (even zeroes) to it **would change it**, making _g_cfg.crc_ from stored config invalid.
So either I'm getting something wrong or I should introduce new g_cfg.version and deal with 'migration' (which looks like a little overkill for such minor change).

openshwprojects, 2023-04-10:
You don't need to change the size of config, there is still a field left here:
[image]
just take 4 bytes from that and add an integer field, it should be ok. It will also work with current CRC calculation, because that field by default will be 0.

openshwprojects, 2023-04-12:
So maybe I will just add it myself for you?

no2chem, 2023-04-14:
So, based on what I discovered in #787 it does seem like PWM frequency makes a considerable difference!

[image]
On the left is a led with 10k PWM, on the right with 25K. Much more pleasant light output.

Some other notes:

It looks like on this led WW and CW are connected to a BP5778EJ driver but RGB are controlled by discrete transistors, maybe they have their own max PWM frequency.. perhaps it would be a good idea to have it configurable per channel.

The BP5778EJ datasheet says 10KHz max PWM but seems to work well above this level.

The brightness and dynamic range seems to be significantly affected by PWM frequency. At brightness=1 the 10KHz light is on but the 25KHz doesn't appear to be visibly on until brightness 15, and even then much much dimmer. Is the brightness controlling the duty cycle, or maybe there's a calculation error...

openshwprojects, 2023-04-14:
So, what do you say, should I add that option tomorrow in the morning? The OP didn't respond so far and the mod is very simple to do.

no2chem, 2023-04-14:
Yeah definitely, I think it would be a good start (would it be per channel?). Maybe a command like setChannelPWMFreq would be better.

Probably should check the effect on brightness as well.

openshwprojects, 2023-04-14:
I was planning to add a global setting. Do lights need a variable frequency per colour?

no2chem, 2023-04-14:
Well, from what I found in the bulb I was using, WW and CW are connected to a BP5778EJ (dual channel driver) but RGB are controlled by discrete transistors. So it is possible you might have a light bulb with a lower max frequency on the RGB channels. Although, on this lightbulb, it appears 25KHz works fine for all channels.

no2chem, 2023-04-14:
Well, looking at the PWM code, its probably a bit of work to make it configurable per channel, so maybe just one for all channels for now.