openshwprojects/OpenBK7231T_App #775: Devices sometimes do not re-connect to wifi

no2chem, 2023-04-10:
**Describe the bug**
I have a fairly large (>200 device) openbk deployment, and the devices are wired to always on. Recently we've had to do some electrical work on the house which involved restarting the network infrastructure. This involves restarting the wifi and MQTT server.

After the WiFi restarts, it seems like some devices do not reconnect to the AP at all. I'm not sure if it's because theres also a short period of time that MQTT is unavailable, or the traffic of 100s of devices trying to rejoin the network at once.

I haven't been able to consistently reproduce the issue, unfortunately. It is a bit of a pain because all the devices are wired to always on, the only way to restart them is to go to the circuit breaker. (feature idea: maybe allow for a special bluetooth debug mode? Don't these devices all have a BT stack too?)

**Firmware:**
Various, but observed on

1.15.542
1.15.580
1.15.601

on N and T devices.

These devices are configured in different ways, switches, lights, etc. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect 100s of openBK devices
2. Restart wifi and MQTT
3. Observe some devices do not come back online

**Additional context**

I'm going to try out the ping watchdog utility, but it doesn't seem to be scriptable, and the ping frequency doesn't seem to be configurable at the moment.

openshwprojects, 2023-04-10:
Hello, what kind of  features do you need when it comes to ping watchdog? I can try  to figure something out for you and provide the required interface.

Those devices do not connect to WIFI? Some log would be helpful.

In code I can see:
[image]
So maybe something like:
addChangeHandler noPingTime > 600 reboot
Maybe we could also add a random value to that somehow...

Why do you need a ping frequency? Is this because frequent pings overload the network? It's very easy to do, I can add it today..

no2chem, 2023-04-10:
Hey, thanks for the quick response.

Yeah, I think 1 second pings is probably too much, that is 300 packets/second of just background traffic. 

They don't connect to wifi, so I can't really see any logs, unfortunately. Maybe if I can reproduce it I can hook a device up to serial and see the logs, but for now I have to restart the device...

I imagine ping frequency is something we don't need to do much testing on (it'll be very quickly evident that the device isn't pinging at the expected frequency).

As for commands, I need some way to set the host to ping in a script or command. 

maybe `PingHost <hostName>`?

openshwprojects, 2023-04-10:
I will commit now PingHost  and PingInterval and we'll think about improving it further.

Do you need ability to change ping host at runtime? Or just from autoexec.bat?

openshwprojects, 2023-04-11:
`addChangeHandler noMQTTTime > 600 reboot `
Ok I am pushing that change now

no2chem, 2023-04-11:
Great, will test out 673 and let you know how it works as soon as I update everything