openshwprojects/OpenBK7231T_App #776: Device does not report Power Factor/Apparent Power/Reactive Power through status command

KaMyKaSii, 2023-04-10:
**Describe the bug**
I was happy to have these readings added to the web interface recently but I would like these same values to be provided by the status command as well.
[image]
**Firmware:**
- Version: 1.15.668
- Device: LS-PA3 (smart plug + power meter)
- Chip/model: BK7231T

openshwprojects, 2023-04-10:
Thank you for reporting, I added missing fields to current build. Can you check latest version?

KaMyKaSii, 2023-04-10:
> Thank you for reporting, I added missing fields to current build. Can you check latest version?

There is a typo in the following line:
https://github.com/openshwprojects/OpenBK7231T_App/blob/15ffc0e16693a8d465daacfeff9c8274b9d95ed8/src/httpserver/json_interface.c#L230

openshwprojects, 2023-04-10:
Thx, fixed

KaMyKaSii, 2023-04-10:
> Thx, fixed

Tested, but there are some issues. The power factor is frozen at 0 even though the web interface works perfectly. I would also like to know if the device itself reports the reactive power or if the firmware does some calculation through the other values, as it is reported up to 12var for a 42W incandescent bulb
[image]

openshwprojects, 2023-04-11:
Thanks, added, there was a stub.

The values are calculated by the snippet submitted by @Diggar in drv_bl_shared.c
[image]

KaMyKaSii, 2023-04-11:
@openshwprojects I noticed that the Power Factor/Apparent Power/Reactive Power readings update in the json response only if the web interface is open. Is it possible to fix this?
Also, in the code the calculation of reactive power is done from the square root of the difference between apparent and active power, but isn't reactive power just the difference between both?

KaMyKaSii, 2023-04-28:
Yes, the formula is correct. The problem is that the active power reported by the device and the calculated apparent power will never be exactly the same even in a purely resistive load. This difference, even if small, will cause the calculation to report reactive power, raising the question of whether it is really valid for the firmware to try to calculate the reactive power.

Diggar, 2023-04-29:
> The problem is that the active power reported by the device and the calculated apparent power will never be exactly the same even in a purely resistive load. This difference, even if small, will cause the calculation to report reactive power

I don't get what you mean. The voltage, current and active power are measured as precise as the chip can measure it and as precise the calibration for all three values was done. Also the calculations are quite precise. Thus, why should they never me measured (and calculated) to the same value for a pure resistive load?
In my opinion it is all correct and the issue can be closed.
By the way: Tasmota does calculate it the same way.