openshwprojects/OpenBK7231T_App #781: non-zero current in the off state

gogo10, 2023-04-12:
**Describe the bug**
Non-zero current in the off state.

**Firmware:**
- Chip/model: BK7231N 
- Energy meter BL0942

[image]

I don't know if this is a normal state or some error, please advise.
Thanks.

openshwprojects, 2023-04-12:
Hello, was the device calibrated?

We could force zero value in off...

gogo10, 2023-04-12:
Yes, it is calibrated, with a load of 60W it shows the correct values.

openshwprojects, 2023-04-12:
I will add a flag to force-zero results when relays are off.

gogo10, 2023-04-14:
I'm sorry, it doesn't work properly, even when switched on and loaded with a 60W bulb, zero values are displayed.

openshwprojects, 2023-04-14:
What is the configuration of your device?

openshwprojects, 2023-04-14:
or which flag does not work properly? 25 or 38?

gogo10, 2023-04-14:
I just wanted to write if the problem is not that I have a bistable relay and in the configuration it is controlled as a bridge?

openshwprojects, 2023-04-14:
This changes much. The current flag only checks relays. So what is your current configuration? Pins and drivers etc

openshwprojects, 2023-04-14:
Ok, can you check now, once the build is finished (wait 10 minutes for Github to process release)