openshwprojects/OpenBK7231T_App #782: Issue with DeviceGroups working intermittently with two Tuya BK7231N smart bulbs

thelinusgo, 2023-04-12:
**Describe the bug**

I've got 2x Tuya `BK7231N` bulb running the latest firmware, and a Tuya Smart Switch `SHWSW2` running Tasmota firmware `12.4.0`, with two switches on it, and I am trying to use it to control each bulb, with a DeviceGroup per bulb / switch.

The light switch is connected to my two bulbs via those two groups, and the main issue seems to be that the command for turning on/off the light via the switch isn't received by the bulbs intermittently. It works sometimes (I would wager a 75-80% chance of it working through the Tasmota light switch), however turning on and off the lights directly works 100% of the time without issue.

I've also checked out both the Tasmota logs and the lightbulb logs at the highest log setting and I have not found anything amiss.

Happy to provide further context or logs if needed!

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure lights are turned off or on
2. Toggle the lights by pressing the button on the light switch
3. Observe that intermittently the lights do not respond to the light switch.

NorthernMan54, 2023-04-23:
I wonder if I have the same issue.   I have a switch running Tasmota 9.5 that I have setup to control my BK7231T led Panels.  And device groups work really well after a reboot, but several hours later it stops working.  A reboot will resolve the issue

I put DGR logs on ExtraDebug and see this

192.168.1.184 is my Tasmota switch

```
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 67
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 68
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 69
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 70
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 71
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 72
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 73
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 74
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 75
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 76
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 77
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 78
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR ignoring message from duplicate or older sequence 79
```

And if I restart my BK7231T led Panels it starts working again, and I get this while turning on and off the Tasmota switch

```
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
Info:DGR:DGR_Parse: [192.168.1.184] seq 0x000B, flags 0x00
Debug:DGR:Next section - 128
Debug:DGR:Power event - values 1, numChannels 1, chans=
Debug:DGR:[ON]
Debug:DGR:
Debug:DGR:Next section - 0
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
Info:DGR:DGR_Parse: [192.168.1.184] seq 0x000C, flags 0x00
Debug:DGR:Next section - 128
Debug:DGR:Power event - values 0, numChannels 1, chans=
Debug:DGR:[OFF]
Debug:DGR:
Debug:DGR:Next section - 0
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
Info:DGR:DGR_Parse: [192.168.1.184] seq 0x000D, flags 0x00
Debug:DGR:Next section - 128
Debug:DGR:Power event - values 1, numChannels 1, chans=
Debug:DGR:[ON]
Debug:DGR:
Debug:DGR:Next section - 0
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
Info:DGR:DGR_Parse: [192.168.1.184] seq 0x000E, flags 0x00
Debug:DGR:Next section - 128
Debug:DGR:Power event - values 0, numChannels 1, chans=
Debug:DGR:[OFF]
Debug:DGR:
Debug:DGR:Next section - 0
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
Info:DGR:DGR_Parse: [192.168.1.184] seq 0x000F, flags 0x00
Debug:DGR:Next section - 128
Debug:DGR:Power event - values 1, numChannels 1, chans=
Debug:DGR:[ON]
Debug:DGR:
Debug:DGR:Next section - 0
ExtraDebug:DGR:Received 26 bytes from 192.168.1.184
Info:DGR:DGR_Parse: grp name test len 4
Info:DGR:DGR_Parse: [192.168.1.184] seq 0x0010, flags 0x00
Debug:DGR:Next section - 128
Debug:DGR:Power event - values 0, numChannels 1, chans=
Debug:DGR:[OFF]
Debug:DGR:
Debug:DGR:Next section - 0

openshwprojects, 2023-04-24:
@NorthernMan54  this is very interesting , I added more debugging, can you recheck and paste the log again, but this time with new debug data included?
Update OBK and check.

I have dgr used for led strips and i haven't observed this problem yet, so any logs from you can be very helpful.

Maybe there is some kind of logical mistake in sequence check

NorthernMan54, 2023-04-24:
LOL, Yesterday I went and applied the almost identical changes in the same area of code, and updated one of my devices. And at the same time I have disabled `IGMP snooping` in my network, thinking it may have been part of it.  And since then I have yet to have the issue re-occur, but previously I did find that it took about a day for the issue to occur.  Will keep monitoring.

In regards to `IGMP Snooping` am thinking this is a contributing factor after reading the description, as it filters multicast packets from devices it thinks are not participating in the multicast.

openshwprojects, 2023-04-24:
maybe the array size is too small, somehow? Maybe we need a debug function to dump all members?

It would make a lot of sense, because in my testing environment, there is not many members, and they surely won't fit whole array, but for other people, on other hand.... it really make sense.... it seems you've discovered a potential cause for the issue! Good job!

NorthernMan54, 2023-04-24:
I'm redoing this with the below, it will be a bit noisy, but should highlight the issue.

```
dgrMember_t *findMember()
{
	int i;
	for (i = 0; i < g_curDGRMembers; i++)
	{
		if (!memcmp(&g_dgrMembers[i].addr, &addr, sizeof(addr)))
		{
			return &g_dgrMembers[i];
		} else {
			addLogAdv(LOG_INFO, LOG_FEATURE_DGR, "g_dgrMembers[i].addr %s != addr %s \n", inet_ntoa(g_dgrMembers[i].addr), inet_ntoa(addr));
		}
	}
	i = g_curDGRMembers;

	addLogAdv(LOG_DEBUG, LOG_FEATURE_DGR, "findMember: %d\n", g_curDGRMembers);
	if (i >= MAX_DGR_MEMBERS)
		return 0;
	g_curDGRMembers++;
	memcpy(&g_dgrMembers[i].addr, &addr, sizeof(addr));
	g_dgrMembers[i].lastSeq = 0;
	return &g_dgrMembers[i];
}
```

NorthernMan54, 2023-04-24:
I have 10 devices sending device group packets, majority are Tasmota

[image]

openshwprojects, 2023-04-24:
or maybe do even  better, just keep IP as byte adr[4] in dgrMember_t and just keep comparing only those 4 bytes.

NorthernMan54, 2023-04-24:
Humm, did I code this wrong....

```
ExtraDebug:DGR:Received 20 bytes from 192.168.1.194
Info:DGR:DGR_Parse: grp name test len 4
Info:DGR:g_dgrMembers[i].addr 16.2.17.95 != addr 16.2.17.95 
```

```
addLogAdv(LOG_INFO, LOG_FEATURE_DGR, "g_dgrMembers[i].addr %s != addr %s \n", inet_ntoa(g_dgrMembers[i].addr), inet_ntoa(addr));
```

openshwprojects, 2023-04-24:
That seem to confirm what I said! Let me get a quick fix for you.

NorthernMan54, 2023-04-24:
Saw my issue, redoing with this

```
dgrMember_t *findMember()
{
	int i;
	for (i = 0; i < g_curDGRMembers; i++)
	{
		if (!memcmp(&g_dgrMembers[i].addr, &addr, sizeof(addr)))
		{
			return &g_dgrMembers[i];
		}
		else
		{
			addLogAdv(LOG_INFO, LOG_FEATURE_DGR, "g_dgrMembers[%d].addr %s != addr %s \n", i, inet_ntoa(g_dgrMembers[i].addr.sin_addr), inet_ntoa(addr.sin_addr));
			addLogAdv(LOG_INFO, LOG_FEATURE_DGR, "g_dgrMembers[%d].addr sin_len %d != %d \n", i, g_dgrMembers[i].addr.sin_len, addr.sin_len);
			addLogAdv(LOG_INFO, LOG_FEATURE_DGR, "g_dgrMembers[%d].addr sin_family %d != %d \n", i, g_dgrMembers[i].addr.sin_family, addr.sin_family);
			addLogAdv(LOG_INFO, LOG_FEATURE_DGR, "g_dgrMembers[%d].addr sin_port %d != %d \n", i, g_dgrMembers[i].addr.sin_port, addr.sin_port);
		}
	}
```

openshwprojects, 2023-04-24:
I'd bet that sin_port changes and that's the cause of random problems on random routers.

NorthernMan54, 2023-04-24:
Just rebuilding with that change.  Will deploy it and let it run.  Will need ~48 hours to say it is resolved, but this looks good.  Will update then

PS This is an awesome package, well done

NorthernMan54, 2023-04-25:
I had my FEIT LED Panels teardown posted back in January.

So I had included some more debugging logic to the code

```
dgrMember_t *findMember()
{
	int i, ip;

	ip = addr.sin_addr.s_addr;

	for (i = 0; i < g_curDGRMembers; i++)
	{
		if (g_dgrMembers[i].ip == ip)
		{
			return &g_dgrMembers[i];
		}
		else
		{
			addLogAdv(LOG_INFO, LOG_FEATURE_DGR, "g_dgrMembers[%d].ip %d != %d \n", i, g_dgrMembers[i].ip, ip);
		}
	}
	i = g_curDGRMembers;
	if (i >= MAX_DGR_MEMBERS)
		return 0;
	g_curDGRMembers++;
	g_dgrMembers[i].ip = ip;
	g_dgrMembers[i].lastSeq = 0;
	return &g_dgrMembers[i];
}
```

And am getting this output, which is not correct, something is reseting the contents of g_dgrMembers

```
nfo:DGR:DGR_Parse: grp name test len 4
Info:DGR:g_dgrMembers[0].ip 0 != -1040078656 
Info:DGR:g_dgrMembers[1].ip 0 != -1040078656 
Info:DGR:g_dgrMembers[2].ip 0 != -1040078656 
Info:DGR:g_dgrMembers[3].ip 0 != -1040078656
```

Looking at the code further I saw this bit

```
void DRV_DGR_Init() {
  memset(&g_dgrMembers[0], 0, sizeof(g_dgrMembers));
#if 0
	DRV_DGR_StartThread();
#else
  DRV_DGR_CreateSocket_Receive();
  DRV_DGR_CreateSocket_Send();
```

Which I think is getting triggered for some reason as this showed up in a log earlier - https://github.com/openshwprojects/OpenBK7231T_App/issues/782#issuecomment-1520435300

```
Info:DGR:DRV_DGR_CreateSocket_Receive: Socket created, waiting for packets
Info:DGR:DRV_DGR_CreateSocket_Send: socket created
```

openshwprojects, 2023-04-25:
Hello, do you think there is still some kind of issue?

DRV_DGR_Init is called by "startDriver DGR", so it seems it is called only once, on wifi connect. Futhermore, "startDriver" will not restart driver if it is already running.

Still, I see that count is not set to 0 there, so I pushed a fix:
https://github.com/openshwprojects/OpenBK7231T_App/commit/d64253c47bf7dfbf551f817eb10ecd9fe7f1fab8
This will help in scenario where memset is called second time, otherwise it would not do a full clear of members (setting count to 0 was missing)

NorthernMan54, 2023-04-25:
Updated to the latest fix including my logging, and will need about 24-48 hours to confirm, but so far it looks promising.  I saw an DRV_DGR_Init occur, and the device groups did **NOT** grow this time.  Fingers crossed on this fix.

```
ug:DGR:DGR_CheckSequence: argument 157, last 0
ExtraDebug:DGR:Seq ok
Info:DGR:DGR_Parse: [192.168.1.194] seq 0x009D, flags 0x00
Debug:DGR:Next section - 128
Debug:DGR:DRV_DGR_processPower: cnt 1, val 1
Debug:DGR:Power event - values 1, numChannels 1, chans=
Debug:DGR:[ON]
Debug:DGR:
Debug:DGR:Next section - 0

Info:DGR:DRV_DGR_CreateSocket_Receive: Socket created, waiting for packets
Info:DGR:DRV_DGR_CreateSocket_Send: socket created

ExtraDebug:DGR:Received 26 bytes from 192.168.1.194
Info:DGR:DGR_Parse: grp name test len 4
ExtraDebug:DGR:DGR_CheckSequence: argument 158, last 0
ExtraDebug:DGR:Seq ok
Info:DGR:DGR_Parse: [192.168.1.194] seq 0x009E, flags 0x00
Debug:DGR:Next section - 128
```

NorthernMan54, 2023-04-26:
24 hours in, and everything is working perfectly

openshwprojects, 2023-04-27:
Very good news! @thelinusgo can you test it as well?

NorthernMan54, 2023-04-27:
48 hours in, and everything is still working perfectly.   I think we nailed it.  You can consider the issue I identified as resolved and closed.

Thanks very much.

thelinusgo, 2023-04-28:
@openshwprojects Apologies, just saw this now. Happy to give it a crack.

thelinusgo, 2023-04-28:
I've flashed these onto my three tuya bulbs and so far so good. I'll report in tomorrow but it seems stable!

thelinusgo, 2023-04-29:
Feel free to close it @openshwprojects. Sometimes there's a delay when I send commands on my Tasmota switches but this existed already.

closed 2023-04-29 by thelinusgo