openshwprojects/OpenBK7231T_App #786: Flag 12 can wear out flash early

no2chem, 2023-04-13:
From what I can tell, flag 12 writes every single state change to flash. This includes changes from DDP.

All these writes to flash might wear out the flash early. I would suggest putting a warning on flag 12 so people know that the flash can wear out prematurely if they enable it.

openshwprojects, 2023-04-14:
Situation is indeed much better than you think. Flashvars structure size is 64 bytes, but we write it gradually in the two sectors (8192 bytes) and do erase only when they are full. So, 8192/64=128 write cycles per single erase.

Still, I will add a warning, thanks.

What else can we do? Force-disable flag 12 when DDP is running?

no2chem, 2023-04-14:
Yeah, I think definitely it should be turned off when DDP gets turned on. Great that it's not 1 write cycle per erase. 

Probably should do some back of the envelope calculations here: 

Typical MCUs have write endurance of 10k cycles or so, my guess is BK devices probably have less due to the price point.

If we do 128 write cycles per erase, we can do 1,280,000 updates before the flash goes poof. Probably fine for manual control.

1 update / day = 35K years
1 update/ hour = 1461 years
1 update / minute = 24.35 years

I think anything less than 1 update per minute, who cares. This includes adaptive lighting, which does around 1 update per minute.

Where it starts getting concerning is protocols like DDP which default to something like 30fps...

@ 1fps= 0.4509 years
@ 5fps = 29 days
@ 30fps = 4.9 days

Of course, this assumes 10K write cycles, and my guess is that probably the BK devices are more like 1K (has anyone tested?)

no2chem, 2023-04-14:
another thought is do BK devices have brown out detector (BOD)? Maybe you can write to flash only when you get brownout, would save lots of flash cycles.

openshwprojects, 2023-04-17:
When in SPI mode, BK7231 identifies as a flash that can be found to have
  Minimum 100K endurance cycle
 Data retention time 20 years
https://www.elektroda.com/rtvforum/topic3931424.html
but still, 10 times more doesn't help much in this scenario.

So, what do you say, should we force-disable the colour save while DDP is running? I don't know about BOD in bk yet.

Maybe we could also alter the save mechanism to save colour/etc only if it doesnt' change for, let's say, one minute?

no2chem, 2023-04-17:
yes, perhaps the function for setting led color should take an optional flag whether to save it or not (and DDP can pass FLAG_NOSAVE, for example).

openshwprojects, 2023-04-29:
led_saveInterval added. Can you check? Of course, it saves only if modified. There is also a check for that in BK hal.