openshwprojects/OpenBK7231T_App #787: Red channel seems to be on even on white mode

no2chem, 2023-04-14:
**Describe the bug**
I've noticed on my some of my recessed lights (Tramsmary 6in recessed bulbs) that the red channel seems to be on in white mode. 

These lights are PWM driven, 5 channels (RGBCW).

I've attached a picture taken with a 1/12000 shutter speed showing the red light on. All my other light bulbs use i2c drivers which seems to have much higher light quality and no banding (maybe related to #774), so I can't tell if the bug is with the hardware or openbk.

[image]

**Firmware:**
- Version 1.15.674
- Device Tramsmart 6in recessed bulb
- Chip/model: BK7231N

**To Reproduce**
Turn light on white
Take picture with high, e.g. 1/12000 shutter speed.

no2chem, 2023-04-14:
Well, actually this might be an issue with the light. I reassigned red to an unused pin and I still saw the red with the banding.

Will try to adjust the PWM frequency to see if it disappears.

openshwprojects, 2023-04-14:
The values of channels should be also shown on GUI, so you can double check if it's something in the channel of the PWM, or a hardware issue.

no2chem, 2023-04-14:
So yeah, I took apart the device and its definitely a hardware issue (it seems like at low frequency the reflector gives a reddish tinge for some reason. It's definitely a lot less in the stock firmware, presumably because of the 10KHz PWM they use. Closing.

closed 2023-04-14 by no2chem