openshwprojects/OpenBK7231T_App #788: AND OR for IF conditions?

ilengyel, 2023-04-14:
Is it possible to have AND OR elements in if conditions? I am trying to write a timer script for some lights:

```
startDriver ntp
ntp_timeZoneOfs 10:00

alias low_lights backlog led_temperature 500; led_dimmer 5; led_enableAll 1; echo lights_low
alias high_lights backlog led_temperature 300; led_dimmer 30; led_enableAll 1; echo lights_high
alias off_lights backlog led_enableAll 0; echo lights_off

addClockEvent 19:00 0xff 1 high_lights
addClockEvent 21:00 0xff 1 low_lights
addClockEvent 23:00 0xff 1 off_lights

delay_s 10

if $hour>=21AND$hour<23 then low_lights
if $hour>=19AND$hour<21 then high_lights
if $hour>=23 then off_lights
if $hour<19 then off_lights
```

I am not seeing any errors in the logs with the above script, but it appears that the conditions `$hour>=21AND$hour<23` and `$hour>=19AND$hour<21` evaluate to true.

openshwprojects, 2023-04-14:
Hello,
I really had to read your script three times to realize, but we obviously use the following syntax:
[image]
So try && and ||

ilengyel, 2023-04-15:
Thanks, that worked, here is an update of my working timer script:

```
startDriver ntp
ntp_timeZoneOfs 10:00

alias high_lights backlog led_temperature 300; led_dimmer 30; led_enableAll 1; echo lights_set_high
alias low_lights backlog led_temperature 500; led_dimmer 5; led_enableAll 1; echo lights_set_low
alias off_lights backlog led_enableAll 0; echo lights_set_off

// Events to turn on, dim and turn off
addClockEvent 18:02 0xff 1 high_lights
addClockEvent 21:00 0xff 1 low_lights
addClockEvent 23:00 0xff 1 off_lights

delay_s 8

// Determine initial state based on time of day
if $hour>=18&&$hour<21 then high_lights
if $hour>=21&&$hour<23 then low_lights
if $hour>=23||$hour<18 then off_lights
```

closed 2023-04-15 by ilengyel

openshwprojects, 2023-04-15:
Can I put that in autoexec examples? Thanks!