openshwprojects/OpenBK7231T_App #790: [BL602] New marking on BL602 chips: LF686C20

ataa, 2023-04-14:
I just received new bulbs from the same seller and same manufacturer, but after opening it I noticed that there's a new chip inside and it's not BL602, since everything else on the module and the power supply were the same and the uart pins were the same and also board rev. was the same, I decided to flash it using BLDevCube and it was a success.
So LF686C20 is a new marking for BL602 (or BL604?)
Not sure what is going on as the chip maker is a different company: https://www.leapfive.com/en/?page_id=4165
Datasheet: https://www.leapfive.com/wp-content/uploads/2020/09/LF686-Datasheet.pdf

[image]

openshwprojects, 2023-04-14:
Thanks, I have already heard something like that, but in general, for such information we have the forum:
https://www.elektroda.com/
If you want, please submit module info + photos + pinout. The forum is well indexed by Google, so people will find your info faster there.

ataa, 2023-04-14:
Ok, Can you add the LF686C20 chip as supported chip to the Documentation/Wiki as well?