openshwprojects/OpenBK7231T_App #793: Requset $NTPOn constant

ostat, 2023-04-18:
**Describe the bug**
Feature request to add a new constant `$NTPOn` to indicate if NTP is configured and initialized.

**Firmware:**
- Version 1.16.9
- Chip/model: BK7231T

**Additional context**
In a script, when checking time variables like `$hour` if the NTP is not configured or the NTP has not initialized an invalid value is returned. For example $hour will be 0 at midnight and on boot before NTP is initialized. This can cause undesired results.
A work around I have been using is to check `if $hour==0&&$minute==0 then goto skip`, but a nicer solution would be to have a variable similar to `$MQTTOn`, to indicate if NTP has been initialized.

openshwprojects, 2023-04-18:
Hello, maybe it would be better to support **waitFor NTPOn 1** or smth like that?

ostat, 2023-04-18:
I like that, should work well. Saves the need to have it retry after a period of time which is what I was doing.

openshwprojects, 2023-04-18:
can you post your script once you get it working so we can attach it to examples?

ostat, 2023-04-18:
Can do

ostat, 2023-04-19:
This is working well. There are a few way to use the changes, you can also use and event or create a loop.
It seemed like the `waitFor NTPState 1` blocking the boot might be causing issues with the logging, but it might be unrelated, still testing that part. I will comment more when I have finished testing.

**Using a wait**
```
startDriver ntp
ntp_timeZoneOfs 10:00

alias day_lights backlog led_temperature 200; led_dimmer 100; echo lights_day
alias night_lights backlog led_temperature 500; led_dimmer 50; echo lights_night

addClockEvent 06:01 0xff 1 day_lights 
addClockEvent 20:01 0xff 1 night_lights 

waitFor NTPState 1
if $hour>=06&&$hour<21 then day_lights
if $hour>=21||$hour<06  then night_lights
```

**Using an EventHandler**
```
startDriver ntp
ntp_timeZoneOfs 10:00

alias day_lights backlog led_temperature 200; led_dimmer 100; echo lights_day
alias night_lights backlog led_temperature 500; led_dimmer 50; echo lights_night

addClockEvent 06:01 0xff 1 day_lights 
addClockEvent 20:01 0xff 1 night_lights 

AddEventHandler NTPState 1 startScript setcolour.bat
```
setcolour.bat
```
if $hour>=06&&$hour<21 then day_lights
if $hour>=21||$hour<06  then night_lights
```

**Using a Loop**
Autoexec.bat
```
startDriver ntp
ntp_timeZoneOfs 10:00

alias day_lights backlog led_temperature 200; led_dimmer 100; echo lights_day
alias night_lights backlog led_temperature 500; led_dimmer 50; echo lights_night

addClockEvent 06:01 0xff 1 day_lights 
addClockEvent 20:01 0xff 1 night_lights 

addRepeatingEvent 2 1 startScript setcolour.bat
```
setcolour.bat
```
if $NTPOn!=1 then goto skip

echo setting colour based on time $day $hour $minute $second
if $hour>=06&&$hour<21 then day_lights
if $hour>=21||$hour<06  then night_lights
goto end

skip:
  echo skipping colour setting, NTP is not set
  addRepeatingEvent 2 1 startScript setcolour.bat

end:
```

openshwprojects, 2023-04-19:
Hello, what do you mean here:
"It seemed like the waitFor NTPState 1 blocking the boot might be causing issues with the logging, but it might be unrelated, still testing that part."

Our script threads are not real threads, nothing is blocked in "waitFor", it just sets a variable and then before continuing script "Thread" it waits for it to change...

openshwprojects, 2023-04-19:
hmm i am not sure, but do you know that you will get random logs skipped if you open two weblog pages in your browser at the same time?

ostat, 2023-04-19:
That was the reason, I didnt realise, checked and had two tabs windows opened. Thanks

openshwprojects, 2023-04-20:
Btw do you know that we support if ...then .... else .. ?

ostat, 2023-04-25:
This is working as expected. I think we can close it now, thanks for your help. Are you ok with my closing it?
There is already an example with this code [automatic_day_and_night_lights.bat](https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/autoexecs/automatic_day_and_night_lights.bat).