openshwprojects/OpenBK7231T_App #794: Identify when light is turned off/on

ostat, 2023-04-18:
**Feature Request**
Is there a way to know if the device was power off and on again quickly?
When resetting a device we need to turn it on and off 3-5 times. This puts the chip in to a reset mode with a fast or slow flash. 
I assume there is a counter somewhere tracking the number of on and off switches. Would it be possible to access that counter?

**Firmware:**
- Version 1.16.9
- Chip/model: BK7231T

**Additional context**
We have lights controlled with a mains switch. On boot they are set to a brightness level that is appropriate for the location.  I would like to create a script that sets the light to the max brightness if it was turned off and on again quickly.

openshwprojects, 2023-04-18:
I can add $failedBoots variable. The reboot is marked as failed if device is up for time shorter than 5 seconds (configurable in general menu)

ostat, 2023-04-19:
I was not able to get the `$failedBoots` to work. I did see that the value is one before the boottime then goes to zero. I tried changing boot time to 10seconds and played with turning on and off the power but I was not able to get failedBoot to go to 2 before the boot time or 1 after the boot time.

```
startDriver ntp
ntp_timeZoneOfs 10:00

echo boot failed boots $failedBoots, uptime $uptime, time $day $hour $minute $second

addRepeatingEvent 1 -1 echo boot failedBoots $failedBoots; uptime $uptime; time $day $hour $minute $second

//Override dynamic temp on fast reboot (mains off and on)
if $failedBoots>=2&&$uptime<5 then goto override 
if $failedBoots>=1&&$uptime>=5 then goto override 
goto continue

override:
echo overriding dynamic light, setting to max brightness, let there be light. failedBoots $failedBoots uptime $uptime
backlog led_temperature 154; led_dimmer 100; led_enableAll 1;
goto end:

continue:
echo setting up dynamic lighting. failedBoots $failedBoots uptime $uptime
backlog led_temperature 500; led_dimmer 20; led_enableAll 1;

end:
```

ostat, 2023-04-19:
Thanks for the update. This is working for me. Of note failedboots is now one higher than you might expect. its number of failed + current boot. 

This is my example boot script (I will come back and edit if I tweak it).

The script will set the light brightness and color dynamically based on the time of day. 
The intention is to have bright lights during the day and dull soft lighting of a night. With the ability use the mains light switch to override to max light if needed.
It uses `waitFor NTPState 1` to ensure that the NTP has set up before setting the initial value at boot time.
Turning off/on the light during the boot success time (5s) will override the dynamic lighting and set to max brightness.

```
startDriver ntp
ntp_timeZoneOfs 10:00

//Override dynamic temp on fast reboot (mains off and on)
//to trigger turn on for less than configured failed boot time(default 5s), off for 2s, on again.
if $failedBoots>=2 then goto override 
goto continue

override:
echo overriding dynamic light, setting to max brightness, let there be light. failedBoots $failedBoots uptime $uptime
backlog led_temperature 154; led_dimmer 100; led_enableAll 1;
goto end

continue:
echo setting up dynamic lighting. failedBoots $failedBoots uptime $uptime

alias day_lights backlog led_temperature 200; led_dimmer 100; echo lights_day_lights
alias night_lights backlog led_temperature 500; led_dimmer 50; echo lights_night_lights

//Timed events for auto switching when leaving the light on
addClockEvent 06:01 0xff 1 day_lights 
addClockEvent 20:01 0xff 1 night_lights 

//set initial value while we wait
backlog led_temperature 500; led_dimmer 20; led_enableAll 1;

//Wait for NTP to set up and set current value
waitFor NTPState 1
if $hour>=06&&$hour<20 then day_lights
if $hour>=20||$hour<06  then night_lights

end:
echo autoexec done. failedBoots $failedBoots, uptime $uptime, NTPOn $NTPOn, time $day $hour $minute $second
```

@openshwprojects, thanks for your support

ostat, 2023-04-25:
This has been working ok for me.
This is an extension to the [automatic_day_and_night_lights.bat] (https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/autoexecs/automatic_day_and_night_lights.bat)

The main difference being, this version supports using the mains switch override the dynamic control. Not sure if you want to extend that one, or create a new example.
Otherwise we can close this one, thanks again.