openshwprojects/OpenBK7231T_App #796: Auto update - Stable/Latest Development

Wolfieeewolf, 2023-04-18:
Is it possible to incorporate an auto-update feature for releases? It would be very helpful if we could choose between stable and development builds and set the frequency of updates (daily, weekly, or monthly). Currently, updating my 20 devices is a tedious process and an automatic update would save me a lot of time.

Another suggestion could be to implement a one-button update feature that would update the device to the latest release version. Perhaps a Tasmo-Admin type interface could be used for this purpose, although I understand that setting it up may require a significant amount of effort.

openshwprojects, 2023-04-19:
I can easily write a script for you for this feature. In fact, there even may be one already on our forum....

https://www.elektroda.com/rtvforum/topic3963037.html

Auto update button is problematic because we don't support HTTPS on OBK to save flash memory.... so we can't directly query github... someone would have to offer a HTTP midleman that automatically fetches releases and hosts them...

Masz updating is usually not recommended, when the device is working well, there is no need to update it manually in most cases. We have some people testing new releases (our new contributor https://www.elektroda.com/rtvforum/uzytkownik3444192.html is also very active when it comes to testing), but we are not a company, so mistakes can slip.

jbornema, 2023-04-19:
Yes - I use curl like this to automatically update:
```
# initiate OTA on device 192.168.0.201
curl --data-binary "@f:\temp\OpenBK7231N_1.17.10.rbl" http://192.168.0.201/api/ota
# after receiving something like {"size":628960}, reboot
curl http://192.168.0.201/cm?cmnd=reboot
```