openshwprojects/OpenBK7231T_App #799: OpenBK7231N uanble to connect

Hotelk52339, 2023-04-20:
After flashing the tuya IR remote to the OpenBK firmware, everything was fine. I was able to connect to the network and set up the device to connect to my router. But the problem is when I sent the login and password to this device, to connect to my router. The wifi network OpenBK disappeared, and the device did not connect to the router.

I tried to restore it to factory settings by holding the button, but the device does not respond.

Can anyone help me with this?

openshwprojects, 2023-04-20:
Hello, first of all, safe mode can be invoked by doing 5 power on/off cycles (disconnecting power). This will return device to AP mode.

Is it your first first obk device? Is your wifi 2.4G?

Hotelk52339, 2023-04-20:
Heloo @openshwprojects 

Thank you for your replay

Yep, safe mode is worked for me, and i reset my device. Now i can connect to the device via router. 
Thank you so much for your help

But i have another issue, i'm not sure that i should post this here, but maybe someone can help me

Unfortunately, I can't set up IR remote (AUBESS), because I just don’t get the codes. When I send commands from the remote, I get this in console:

Info:MQTT:Publishing val {"IrReceived":{"Protocol":"UNKNOWN","Bits":32,"Data":"0xD68EB9B2"}} to obk1FCB04B9/RESULT retain=0

How can I use this code, to add the integration to Home Assistant?

Thanks in advance

openshwprojects, 2023-04-21:
What kind of remote do you have?

Hotelk52339, 2023-04-21:
@openshwprojects 

I have Aubess Wifi Smart IR Controller

Here is forum about it: https://www.elektroda.com/rtvforum/viewtopic.php?p=20549257#20549257

closed 2023-04-21 by Hotelk52339