openshwprojects/OpenBK7231T_App #800: Setting NTP server and getting date/time/server info

davoau, 2023-04-21:
Hi,
I am trying to set the ntp server address and timezone and I am guessing I require this this for correct energy metering as it does on tasmota - todays usage resets to zero at midnight.

I have tried the below commands to set ntp info. - I have an ntp server running on 10.0.2.1 / ntp.intranet.lan, and tried to set it both with ip address and domain name.

```
mosquitto_pub -t "obkFFD00991/ntp_setServer/set" -m "10.0.2.1"
mosquitto_pub -t "obkFFD00991/ntp_setServer/set" -m "ntp.intranet.lan"
mosquitto_pub -t "obkFFD00991/ntp_timeZoneOfs/set" -m "+10:00"
```
But if I try to retrieve the info all I get is 10, which I guess is the +10:00 of timesone

the below mqtt output is all that I see.
```
obkFFD00991/ntp_info/get
obkFFD00991/0/get 10
```

**Firmware:**
- Build on Apr 19 2023 12:34:41 version 1.17.11
- Aubess 16A Tuya Smart WiFi Switch with Power Metering  
- Chip/model: BK7231N
- P8 - LED_n, P23 Btn, P24 TglChanOnTgl, 26 REL
- Startup Command: startDriver BL0942

Everything else seems to work as expected...

openshwprojects, 2023-04-21:
Hello,
I think that obkFFD00991/ntp_setServer/set syntax is not correct, this is for channels.
For commands, we do:  cmnd/obksmth/ntp_setServer with payload server IP (try IP first).

Btw, why don't you try commands first on the WWW panel, and then check with MQTT?

davoau, 2023-04-21:
Hi, I did try cmnd/ but that made no difference - would not work.
If I use:
` mosquitto_pub -t "obkFFD00991/1/set" -m "1"`
it will turn the switch on and 
`mosquitto_pub -t "obkFFD00991/1/set" -m "0"`
 will turn it off.

If I input the command ntp_timeZoneOfs +10:00 using WEB APPLICATION / LOGS / COMMANDS box at bottom of page I see in logs:
`Error:CMD:cmd ntp_timeZoneOfs NOT found (args +10:00)
Info:CMD:[WebApp Cmd 'ntp_timeZoneOfs +10:00' Result] Unknown command`

I also tried ntp_timeZoneOfs "+10:00"

I have also noticed that if the smart switch is turned off (disconnected from power), then turned on again it does not send any mqtt messages and I am unable to connect to it using a browser until I push the button on the switch.

openshwprojects, 2023-04-22:
All your queries are still incorrect.

For channels (switch) this is correct:

`obkFFD00991/1/set`

I said, for channels. And only for channels. So when you do:
[image]
it tries to parse value as integer and sets channel 0 to 10... it's not a command.

For commands, you do this:

`cmnd/obkFFD00991/ntp_timeZoneOfs`
 with payload 10 or anything

https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/mqttTopics.md

> I have also noticed that if the smart switch is turned off (disconnected from power), then turned on again it does not send any mqtt messages and I am unable to connect to it using a browser until I push the physical button on the switch.
I don't know about the connect issue, it seems strange, but for MQTT, we have flag "broadcast self state on mqtt connect"

davoau, 2023-04-23:
Thanks for the info above.
I assumed NTP would be loaded by default.

After loading NTP driver I set the ntp server and time offset
```
root@pi1:~# mosquitto_pub -t "cmnd/obkFFD00991/ntp_setServer" -m "ntp.intranet.lan"
root@pi1:~# mosquitto_pub -t "cmnd/obkFFD00991/ntp_timeZoneOfs" -m "+10:00"
```
what i see in webapp log
```
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/obkFFD00991/ntp_info
Info:NTP:Server=ntp.intranet.lan, Time offset=36000
```
And on front page of switch it says:
`NTP: Syncing with ntp.intranet.lan`

So it looks like it is all working.
However I noticed there is no energy consumption today, only energy total, so I will have to do the rollover with homeassistant.

I will investigate the problem with no mqtt messages until button push. I had the same problem this morning when I turned on the switch.
However after playing around with the NTP stuff and then restarting (only switched off for 30 seconds), it connected, and started sending mqtt messages to broker, and I could issue the command over mqtt to turn on switch and it worked. Will leave off overnight again and see what happens.

Anyways thanks for all the help.