openshwprojects/OpenBK7231T_App #801: Problem with IR

Hotelk52339, 2023-04-21:
**Firmware:**
- Version [1.17.16]
- Device: AUBESS SMART IR CONTROLLER
- Chip/model: BK7231N CBU

Hello

I have a problem with AUBESS IR. I can read commands from my TV, but when I send the read command, nothing happens. (And this problem is not only with the TV)

Command read by AUBESS:

`Info:IR:IR IR_Sony 0x1 0x15 1`

The command that I send in the console:

`IRSEND Sony 0x1 0x15 1`

But unfortunately, my TV does not react. I also tried to use other devices, and the result was the same everywhere, except for one device - heater

The command that is read by AUBESS for the heater:

`Info:IR:IR IR_NEC 0x80 0x8 0`

The command that I send in the console:

`IRSEND NEC 0x80 0x8 0`

As far as I understood, this one worked because this command has "NEC" prefix. If the prefix is different, then AUBESS does not send commands

My question is, this firmware can only work with NEC? Or is it a problem on my side

openshwprojects, 2023-04-22:
Hello. The Syntax you are using is incorrect. When you look at the docs:
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/commands.md
You can find:
[image]
Please try with correct syntax first.

Hmm, maybe we will add support for your syntax as well, you're not the first person to confuse it...

Hotelk52339, 2023-04-22:
So i tried with new syntax:

`irsend SONY-1-15-1`

Replay from console:

`Info:IR:Delay 45ms`
`Info:IR:Delay 100ms`
 `Info:IR:IR send Sony-1-15-1 protocol 21 addr 0x1 cmd 0x15 repeats 1`

But TV doesn't react on this.

As far as I know, you are currently testing beta firmware, where a lot of other devices have been added to read IR. 
If you want, I can help you to test them

froid-san, 2023-08-07:
Hello, I'm also having the same issue with the exact same device.

Version [1.17.205]
Device: AUBESS SMART IR CONTROLLER
Chip/model: BK7231N CBU

Module Config:
P7 (PWM1):	IRSend
P8 (PWM2):	IRRecv
P24 (PWM4):	WifiLED_n

Get Sony TV Power ON button
Logs:
Info:IR:IR IR_Sony 0x1 0x15 0
Info:IR:IR IR_Sony 0x1 0x15 1
Info:IR:IR IR_Sony 0x1 0x15 1
Info:IR:IR IR_Sony 0x1 0x15 1
Info:IR:IR IR_Sony 0x1 0x15 1

Execute Custom Command:
IRSend Sony 0x1 0x15 1

log:
Info:IR:Delay 45ms
Info:IR:Delay 100ms
Info:IR:IR send Sony 0x1 0x15 1 protocol 21 addr 0x1 cmd 0x15 repeats 

IR light Flashes on Android Camera but TV does not Power ON

Also tried.
IRSend Sony-1-15-1

log:
Info:IR:Delay 44ms
Info:IR:Delay 100ms
Info:IR:IR send Sony-1-15-1 protocol 21 addr 0x1 cmd 0x15 repeats 1
OK

Same result, IR light Flashes on Android Camera but TV does not Power ON