openshwprojects/OpenBK7231T_App #802: add_temperature adds double the expected value

ostat, 2023-04-22:
**Describe the bug**
When using the add_temperature command it appears to add double the expected value.

**Firmware:**
- Version 1.17.14
- Chip/model: BK7231T

**To Reproduce**
Steps to reproduce the behavior:
1. Go to **logs** on **web application**
2. Enter command `led_temperature 300`
3. Enter `echo $led_temperature`, note that is is 300
4. Enter command `add_temperature 50`
5. Enter `echo $led_temperature`, note that is is 400, expected 350

**Screenshots**
[image]

**Additional context**
I think this might be an issue in function **LED_AddTemperature** in file https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/cmnds/cmd_newLEDDriver.c

In **LED_AddTemperature** the command `cur += iVal;` is run multiple times in the one flow.
```c
void LED_AddTemperature(int iVal, int wrapAroundInsteadOfClamp) {
	float cur;

	cur = led_temperature_current;
        
       cur += iVal; //adds increment to current

	if (wrapAroundInsteadOfClamp == 2) {
		// Ping pong mode
		cur += iVal * temperature_pingPong_direction; 
		if (cur >= led_temperature_max) {
			cur = led_temperature_max;
			temperature_pingPong_direction *= -1;
		}
		if (cur <= led_temperature_min) {
			cur = led_temperature_min;
			temperature_pingPong_direction *= -1;
		}
	}
	else if (wrapAroundInsteadOfClamp == 0) {
		cur += iVal;  //adds again here 
		// clamp
		if (cur < led_temperature_min)
			cur = led_temperature_min;
		if (cur > led_temperature_max)
			cur = led_temperature_max;
	}
	else {
		// wrap, but first always reach max/min.
		// So, if we are at min/max then wrap around, otherwise add and clamp...
		if (cur <= led_temperature_min && iVal < 0) {
			// wrapping around from min to max
			cur = led_temperature_max;
		}
		else if (cur >= led_temperature_max && iVal > 0) {
			// wrapping around from max to min
			cur = led_temperature_min;
		}
		else {
			cur += iVal; //adds again here 
			// clamp below is CORRECT. We want it to stop at max/min
			if (cur < led_temperature_min)
				cur = led_temperature_min;
			if (cur > led_temperature_max)
				cur = led_temperature_max;
		}
	}

	LED_SetTemperature(cur, true);
}
```

openshwprojects, 2023-04-22:
You are correct. Good spotting. I don't know how it got there.

Fixed: https://github.com/openshwprojects/OpenBK7231T_App/commit/ca3cec6decaffbf16a028e465ed0730a85dc0785 

Can you check is it working for you now?

ostat, 2023-04-23:
Tested, look good thanks.

ostat, 2023-04-23:
resolved

closed 2023-04-23 by ostat