openshwprojects/OpenBK7231T_App #805: HA MQTT discovery topic is not sending daily energy stats

fokcuk, 2023-04-24:
As per the header - HA MQTT discovery topic is not sending daily energy stats
I can see that its sending MQTT data for daily, hourly etc in the main device topics, but there is no configuration to show what is actually getting sent to the Homeassistant discovery.
When I checked MQTT data, OpenBK is sending hourly stats, but not daily stats. So HA never discovers it
Is there a way to check/change what is getting to sent via HA discovery?

openshwprojects, 2023-04-29:
Thanks. Json format was wrong. It should be ok now. Can you retest?

stefan064, 2024-01-22:
I've noticed this also. When NTP is enabled then `energycounter_yesterday`, `energycounter_today `and `energycounter_clear_date` are all published on MQTT as expected and also show up on the web UI. However none of these are discovered in home assistant, which is because command `scheduleHADiscovery` doesn't include them in the MQTT `homeassistant/sensor` topic.

I tried manually publishing one of them and that fixed it: i.e. after some ~half-hour home assistant discovered the sensor and started showing its value.

Possible cause:
In hass.c -> hass_init_power_sensor_device_info() it looks like `OBK_CONSUMPTION_YESTERDAY, OBK_CONSUMPTION_TODAY, OBK_CONSUMPTION_CLEAR_DATE` are never included and therefore left out. Perhaps these are meant to be conditionally included when NTP is enabled or similar?

stefan064, 2024-02-10:
Here's my attempt at improving this: https://github.com/stefan064/OpenBK7231T_App
- added energycounter_yesterday, energycounter_today, energycounter_clear_date to home assistant discovery
- added 'friendly names' to the energy sensors (i.e. "energycounter_today" ->  "Energy Today")
- adjusted formatting of energycounter_clear_date so home assistant can interpret as a 'timestamp'.
Testing on a few BK7231N devices now with BL0937 / BL0942 energy monitoring chips

[image]