openshwprojects/OpenBK7231T_App #810: Tasmota topics publish interval

serek4, 2023-04-26:
**Describe the bug**

VCPPublishIntervals command doesn't limit mqtt messages published to Tasmota topics.

**Firmware:**
- Version 1.17.29
- Chip/model: BK7231N
- BL0937 driver
- startup command: backlog startDriver NTP; ntp_timeZoneOfs 2; SetupEnergyStats 1 60 60;

Steps to reproduce the behavior:
1. enable: Flag 30 - [MQTT] Enable Tasmota TELE etc publishes (for ioBroker etc)
3. send command: VCPPublishIntervals 30 60
4. tele/# topics are still published in ~3s intervals

openshwprojects, 2023-04-26:
You are correct, but we will need another command for tele.

What about adding TelePeriod command? I can add this tomorrow morning.

serek4, 2023-04-27:
Are you thinking of the same implementation as VCPPublishIntervals ie. `TelePeriod [minInterval] [forcedInterval]`, or just simple `TelePeriod [interval]`?

openshwprojects, 2023-04-27:
What would fit your need better?
From what I can see currently, we do two types of TELE:
- tele/ sensor, done every 3 seconds if sensor is running (any sensor)
- tele / state, done every 120 seconds
[image]
What kind of approach would you like use to take with tele? 

Maybe do not use name "TelePeriod" because it may confuse people.
Maybe TTP (TasTelePeriod) with two arguments - first argument interval for tele/sensor, second for tele/state?

Or should be publish sensor only when it changes?

serek4, 2023-04-27:
> Maybe TTP (TasTelePeriod) with two arguments - first argument interval for tele/sensor, second for tele/state?

That sound good

> Or should be publish sensor only when it changes?

Maybe add command similar to VCPPublishThreshold?

openshwprojects, 2023-04-27:
Well, if you want a command similiar to VCPPublishThreshold, then maybe I will just plug the TELE SENSOR into the VCPPublishThreshold code block? That way VCPPublishThreshold will work for TELE.

The only issue is that I will have to add some other condition for sensors like DHT11,, but I can do that..

serek4, 2023-04-27:
Let's stick with TELE publish in intervals, adjustable by your proposed TasTelePeriod command.

openshwprojects, 2023-04-27:
Pushed: 
TasTeleInterval [SensorInterval] [StateInterval]
Can you check?
It should be put in autoexec or startup command

serek4, 2023-04-27:
It works but I found 2 bugs
1. tele/sensor is additionally sent with tele/status interval. (`MQTT_BroadcastTasmotaTeleSENSOR()` is called  inside `MQTT_BroadcastTasmotaTeleSTATE()` function)
2. flag 2 (Broadcast self state every N seconds) also publish both tele topics with its own interval

openshwprojects, 2023-04-29:
I assume it's solved now, right?

closed 2023-04-29 by serek4