openshwprojects/OpenBK7231T_App #811: How to export configuration as in Tasmota

nospam2k, 2023-04-26:
I'm sorry to ask a noob question, but I don't want to screw up my bulb. I have 9 Daybetter bulbs and I have got one working just as I want. How do I back up the configuration so I can import it into a new bulb?

I want to backup and restore the configuration as it works with Tasmota.

Thanks again. Really cool project! I'm very grateful!

openshwprojects, 2023-04-27:
You can export config JSON in Web App -> Config and import it in Web App ->Import, you can also experiment with binary data dump in Web App->Flash (but that's for advanced users).  Please try on a dummy device first if you are not sure.

nospam2k, 2023-04-28:
Sorry for my confusion. I saw that page that doesn't seem to save all the various configuration parameters such as MQTT, startup flag, etc. I want to back up the whole thing and upload it to another device having to change only the name of the device and update the mqtt topics.

openshwprojects, 2023-04-28:
If you want everything saved, then  Web App -> Flash tab allows you to download binary dump of obk configuration memory.

nospam2k, 2023-04-28:
Is it safe to upload the binary dump to another device (won't overwrite mac or special device specific params)?

openshwprojects, 2023-04-29:
It is safe to upload a config dump from Web App -> Flash tab to another device. MAC is stored elsewhere, it's in the RF partition.

It is not safe to upload whole 2MB dump from another device to other device via BK7231 GUI flash tool.

nospam2k, 2023-04-29:
Ok, thx!

closed 2023-04-29 by nospam2k